1mckenna/esp32_iGrill #1: Support for iGrill v3

christiaaneggink, 2021-01-26:
Awesome @1mckenna, I had given up hope to ever get this working.

For those interested, I got this working for the iGrill v3 by changing the SERVICE_UUID in line 16:
static BLEUUID SERVICE_UUID("6E910000-58DC-41C7-943F-518B278CEA88"); //iGrillv3 Service

1mckenna, 2021-01-26:
Thanks for letting me know! Once I finish up the last bit of changes for the wifi/mqtt side I'll add this in so it'll work for either device without any code changes.

1mckenna, 2021-02-03:
@christiaaneggink iGrillv3 support has been added. When you have some free time can you flash the new codebase and let me know if you run into any issues.

christiaaneggink, 2021-02-05:
@1mckenna, I tried flashing it to my ESP32 Dev Kit V1 yesterday, but the sketch appears to be too big for my board. What board are you using? Did you change partition table sizes?

`Sketch uses 1483773 bytes (113%) of program storage space. Maximum is 1310720 bytes.`

1mckenna, 2021-02-05:
Awesome! 

The probes according to the Weber docs  can read temps from -50* C to 380* C. I have my device set to use Fahrenheit though and never tried in Celsius. I'll change my device today and see what readings I get on my side.

Does the value being published to MQTT match what's shown on the display?  I set the unit_of_measurement to degree F in the probes mqtt autodiscovery configs (see line 916) and don't think that would affect the temp shown (unless maybe HA is doing some temp conversion automatically?). I'll push an update later today to make that easily configurable for those that have the device reading in Celsius in the config.h.

Might also want to make sure the probes are reading the correct temps as I've had a probe go bad before. Easiest way to check is to boil a cup of water and put the probe in it since it should read ~100* C/212* F.

christiaaneggink, 2021-02-05:
Setting the unit_of_measurement to degrees C for all probes did the trick. Home Assistant indeed did auto conversion of the published temperatures and interpreted the published value as degrees F, although it was degrees C :-)

1mckenna, 2021-02-06:
I just pushed new code that enables the ability to select metric degrees from the config portal. 
Since I added a new setting that wasn't there previously after flashing you will need to put the device into configuration mode and then open the portal and re-enter all the information **(even if the old data is shown)** and make the degree selection before saving.

christiaaneggink, 2021-02-08:
Excellent, thank you.

Just out of curiosity, how often does your v2 publish its battery level? I have been testing the v3 for a while now, but have not yet seen an update on the battery level.

1mckenna, 2021-02-08:
The way the code works is once we connect to the iGrill Device via BLE we then register to be notified via our callback once the  batteryCharacteristic changes (i.e. if the battery level changes on the device we get notified of its new value) on notification of the change we then publish the battery level to MQTT. 

I usually see a battery message every few minutes, if you leave it up and running for ~5 mins I would suspect you should see at least one battery level message being published.

closed 2021-02-10 by 1mckenna