1mckenna/esp32_iGrill #2: WiFi configuration not stored

dev-strom, 2021-03-22:
Having troubles to configure the wifi network. The credentials are not stored to flash and got lost after a reboot or reset of the device.
I figured out, that the function `saveConfigData();` will only be called if both ssid and ssid1 are given:

`if ( String(ESP_wifiManager.getSSID(0)) != "" && String(ESP_wifiManager.getSSID(1)) != "" )`

The workaround would be to always set a value to ssid1. Although I don't need it.

But unfortunately this solves only the storage of the credentials. But the device still doesn't connect to wifi after reboot or reset and always enters first the configuration mode.

1mckenna, 2021-03-23:
What ESP board are you using?
What version of ESP_WifiManager libraries are you using?
Can you change _WIFIMGR_LOGLEVEL_ in config.h to 3 then open serial monitor after flashing and post the output (remove your wifi password before posting if its shown)?

dev-strom, 2021-03-23:
I'm using a NodeMCU ESP-325 v1.1 (ESP-WROOM-32).

I changed this in config.h:
```
#define _WIFIMGR_LOGLEVEL_    3
#define WIFIMGR_LOGLEVEL_    3
#define WIFIMGR_LOGLEVEL    3
```

And here is the log after a fresh compile and reboot:

```
s Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[II] Starting iGrill BLE Client using LittleFS on ESP32_DEV
[II] ESP_WiFiManager v1.4.3
[II] ESP_DoubleResetDetector v1.1.1
{"MQTT_SERVER":"server","MQTT_SERVERPORT":"1883","MQTT_USERNAME":"user","MQTT_PASSWORD":"password","MQTT_BASETOPIC":"homeassistant","METRIC_DEGREES":true}[II] 
Config File successfully parsed
{"MQTT_SERVER":"server","MQTT_SERVERPORT":"1883","MQTT_USERNAME":"user","MQTT_PASSWORD":"password","MQTT_BASETOPIC":"homeassistant","METRIC_DEGREES":true}[II] 
Config File successfully parsed
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 10.1.3.27 , gatewayIP = 10.1.3.1
[WM] netMask = 255.255.255.0
[II] ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Additional SSID =  myssid , PW =  password
[WM] Connecting MultiWifi...
```

and after a while it repeats:

```
[WM] WiFi not connected
[II] Not connected to an iGrill Device, Skipping connecting to MQTT
[II] WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Additional SSID =  myssid , PW =  password
[WM] Connecting MultiWifi...
```

The stationIP 10.1.3.27 is correct assigned via dhcp!

And if I start the configuration portal (double reset) and hit _Exit Portal_ the device will connect to my wifi until next reboot or reset.

dev-strom, 2021-03-23:
Oh.
I switched the board configuration from _ESP32 Dev Module_ to _NodeMCU-32S_.
And now the compiler says:

> Sketch uses 1490753 bytes (113%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 52132 bytes (15%) of dynamic memory, leaving 275548 bytes for local variables. Maximum is 327680 bytes.
> Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
> Compilation error: Error: 2 UNKNOWN: text section exceeds available space in board
> 

Maybe this is the problem!?
But the board is labeled with `ESP-325` and not `ESP-32S`!
I don't know what's wrong...

1mckenna, 2021-03-24:
The size too big error is likely because the partition size wasn't updated when you changed board types. Try selection the option shown [here](https://github.com/1mckenna/esp32_iGrill#select-the-correct-partition-scheme) and then see if the log looks any different.

Can you paste a link to the exact board you bought that might help determine the correct options for you to select?

Do you have a spare board you can try flashing this on to see if the issue is related to the board itself? Also if you haven't already you may want to try moving the device closer to the access point to see if its a wifi range/interference issue. 

Can you try one of the ESP_WifiManager examples included with the library to see if you have the same issue getting it to connect and stay connected?

dev-strom, 2021-03-24:
I tried upgrading the ESP_WifiManager to 1.5.0 - no success
I tried downgrading Arduino IDE from 2.0.0-beta3 to 1.8.13 - no success
I tried a different board [DOIT Esp32 DevKit v1](https://docs.zerynth.com/latest/reference/boards/doit_esp32/docs/) - no success

In general wifi works! But it connects only after first starting the configuration portal. Not directly after power on!

I'll try the ESP_WifiManager examples, but it could take a few days... (I don't have the time to do it now.)

1mckenna, 2021-03-24:
Sorry I was thinking the issue was with wifi and not the device going into config mode.

Does this happen if you power the device from something other than a computer? I know on my board after flashing it always enters the portal and I have to press reset to reboot the device, but that behavior doesn't happen if I power from an outlet.

Can you try changing line 32 in config.h to shorten the time it will look for a second reset button push before entering the portal to see if that has any effect.
i.e 
`#define DRD_TIMEOUT 1` 

The default value in the ESP_Wifimanager examples has the timeout set to 10 might want to try value as well if 1 doesn't seem to help.

dev-strom, 2021-03-26:
Last night I had a vision! ;-)

I switched the wifi configuration to a different network. And voilá: all works fine!
I switched back to the previous network and had the same problems as before.

Both wifi networks are hosted by the same access point. The only difference is: the problem wifi has a hidden ssid! I assume, this is the problem. ESP_WifiManager has problems with hidden ssid's.

1mckenna, 2021-03-30:
Glad you got it working! I'll update the README to mention the issues with using a hidden SSID.

closed 2021-03-30 by 1mckenna