1mckenna/esp32_iGrill #6: Support for iGrill mini

jysaloma, 2021-05-30:
Please add support for iGrill mini.

here is the SERVICE UUID:
static BLEUUID MINI_SERVICE_UUID("63C70000-4A82-4261-95FF-92CF32477861"); //iGrill mini Service

I am able to to connect the device but it will crash the ESP32-WROOM-32U in App authentication phase?
...
09:51:25.740 -> [WM] WiFi connected
09:51:25.740 -> [WM] SSID: IOT ,RSSI= -53
09:51:25.740 -> [WM] Channel: 11 ,IP address: 10.0.0.140
09:51:26.419 -> [II] Not connected to an iGrill Device, Skipping connecting to MQTT
09:51:26.554 -> [II] Connecting to iGrill Device: d4:81:ca:00:a5:ab
09:51:27.631 -> [II]  - Created client
09:51:27.631 -> [II]  - Connected to iGrill BLE Server
09:51:27.765 -> [II]  - iGrill Pair Status: Paired
09:51:28.649 -> [II]  - Performing iGrill App Authentication
09:51:28.885 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
09:51:28.919 -> Debug exception reason: Stack canary watchpoint triggered (BTU_TASK) 
09:51:28.919 -> Core  0 register dump:
09:51:28.919 -> PC      : 0x4009202b  PS      : 0x00060c36  A0      : 0x400848bc  A1      : 0x3ffeb1e0  
09:51:28.919 -> A2      : 0x3fff5b58  A3      : 0x00000000  A4      : 0x00000268  A5      : 0x3fff5ca8  
09:51:28.919 -> A6      : 0x3ffeb260  A7      : 0x00000026  A8      : 0x800837b1  A9      : 0x3ffeb230  
09:51:28.953 -> A10     : 0x3fff5b58  A11     : 0x3ffb6cb0  A12     : 0x00000001  A13     : 0x00000000  
09:51:28.953 -> A14     : 0x00001800  A15     : 0x00000004  SAR     : 0x00000002  EXCCAUSE: 0x00000001  
09:51:28.953 -> EXCVADDR: 0x00000000  LBEG    : 0x400900f0  LEND    : 0x400900fb  LCOUNT  : 0x00000010  
09:51:28.953 -> 
09:51:28.953 -> 
09:51:28.953 -> Backtrace:0x40092028:0x3ffeb1e00x400848b9:0x3ffeb2b0 0x4012c257:0x3ffeb2e0 0x4015cc7f:0x3ffeb310 0x40182bdf:0x3ffeb5a0 0x4015dd56:0x3ffeb830 0x401826da:0x3ffeb850 0x4015dd56:0x3ffeb8c0 0x401823cd:0x3ffeb8e0 0x4015cfb9:0x3ffeb900 0x4015d031:0x3ffebb90 0x4015d1f7:0x3ffebbb0 0x40146305:0x3ffebbd0 0x401410f9:0x3ffebe70 0x401415da:0x3ffec100 0x40141f76:0x3ffec170 0x4014344f:0x3ffec1a0 0x4014348e:0x3ffec1c0 0x4014f3ad:0x3ffec1e0 0x4013ef66:0x3ffec360 0x40159873:0x3ffec380 
09:51:28.987 -> 
09:51:28.987 -> 
09:51:28.987 -> 
09:51:28.987 -> 
09:51:28.987 -> ELF file SHA256: 0000000000000000
09:51:29.021 -> 
09:51:29.021 -> Rebooting...
09:51:29.021 -> ets Jun  8 2016 00:22:57

let me know what information is neede to debug more?

1mckenna, 2021-06-12:
Sorry for the delay, you'll want to follow the steps outlined [here](https://github.com/me-no-dev/EspExceptionDecoder) to install the ESPExceptionDecoder so we can get info about whats causing the crash. 

Just based on where its failing looks to me like the AUTH SERVICE UUID isn't the same for the mini.

Since I don't have an iGrill Mini device I'll need to see if you can determine what characteristics we'll need to use. 
The easiest way to do that will be to

1. Install the Weber iGrill App and pair the device to the phone via the app
2. Install the nrfConnect Android App
3. Once in the App Click on the Bonded Tab
4. You should see your iGrill Mini listed, click Connect
5. It should then go through discovery of the services
6. In the listed services can you take a screenshot or attach all the UUIDs shown. 
7. Dont forget to unpair the iGrill device from your phone so you dont run into connectivity issues with the ESP32
8. Once we have all the correct UUIDs for the mini device it should be a simple to get it working.

jysaloma, 2021-06-13:
[
[igrill-mini-nrfConnect-log.txt](https://github.com/1mckenna/esp32_iGrill/files/6643378/igrill-mini-nrfConnect-log.txt)
](url)

jysaloma, 2021-06-13:
10:03:40.079 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
10:03:40.079 -> Debug exception reason: Stack canary watchpoint triggered (BTU_TASK) 
10:03:40.079 -> Core  0 register dump:
10:03:40.079 -> PC      : 0x400930c7  PS      : 0x00060636  A0      : 0x8015d594  A1      : 0x3ffeb100  
10:03:40.116 -> A2      : 0x3ffe54e0  A3      : 0xffffffff  A4      : 0x00000001  A5      : 0x00000000  
10:03:40.116 -> A6      : 0x00001800  A7      : 0x00000004  A8      : 0x800836f7  A9      : 0x3ffeb0f0  
10:03:40.116 -> A10     : 0x3fff5fd8  A11     : 0x00000008  A12     : 0x3ffc2c58  A13     : 0x00000000  
10:03:40.116 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000001  
10:03:40.116 -> EXCVADDR: 0x00000000  LBEG    : 0x4008ff88  LEND    : 0x4008ff9e  LCOUNT  : 0xffffffff  
10:03:40.153 -> 
10:03:40.153 -> 
10:03:40.153 -> Backtrace:0x400930c4:0x3ffeb1000x4015d591:0x3ffeb140 0x4015c6d2:0x3ffeb160 0x4015d509:0x3ffeb180 0x4015b44e:0x3ffeb1a0 0x4012fc2b:0x3ffeb1d0 0x4016065b:0x3ffeb200 0x401865bb:0x3ffeb490 0x40161732:0x3ffeb720 0x401860b6:0x3ffeb740 0x40161732:0x3ffeb7b0 0x40185da9:0x3ffeb7d0 0x40160995:0x3ffeb7f0 0x40160a0d:0x3ffeba80 0x40160bd3:0x3ffebaa0 0x40149cd9:0x3ffebac0 0x40144acd:0x3ffebd60 0x40144fae:0x3ffebff0 0x4014594a:0x3ffec060 0x40146e23:0x3ffec090 0x40146e62:0x3ffec0b0 0x40153f91:0x3ffec0d0 0x4014293a:0x3ffec250 0x4015d24f:0x3ffec270 
10:03:40.191 -> 

------
PC: 0x400930c7: xQueueSemaphoreTake at ../esp-idf/components/freertos/queue.c line 1441
EXCVADDR: 0x00000000

Decoding stack results
0x400930c4: xQueueSemaphoreTake at ../esp-idf/components/freertos/queue.c line 1431
0x4015d591: btc_init at ../esp-idf/components/bt/common/btc/core/btc_task.c line 364
0x4015c6d2: smp_sm_event at ../esp-idf/components/bt/host/bluedroid/stack/smp/smp_main.c line 773
0x4015d509: btc_init at ../esp-idf/components/bt/common/btc/core/btc_task.c line 341
0x4015b44e: smp_calculate_comfirm at ../esp-idf/components/bt/host/bluedroid/stack/smp/smp_keys.c line 677
0x4012fc2b: btc_storage_add_ble_local_key at ../esp-idf/components/bt/host/bluedroid/btc/core/btc_ble_storage.c line 309
0x4016065b: bta_dm_co_ble_set_init_key_req at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_co.c line 397
0x401865bb: bta_dm_acl_change at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c line 3521
0x40161732: bta_dm_get_av_count at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_pm.c line 1118
0x401860b6: bta_dm_search_start at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c line 1498
0x40161732: bta_dm_get_av_count at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_pm.c line 1118
0x40185da9: bta_dm_sdp_result at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c line 3880
0x40160995: bta_dm_pm_timer_cback at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_pm.c line 875
0x40160a0d: bta_dm_pm_stop_timer_by_index at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_pm.c line 297
0x40160bd3: bta_dm_pm_set_mode at ../esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_pm.c line 1139
0x40149cd9: gatts_process_read_by_type_req at ../esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_sr.c line 997
0x40144fae: gap_read_attr_value at ../esp-idf/components/bt/host/bluedroid/stack/gap/gap_ble.c line 222
0x4014594a: GATTS_AddCharDescriptor at ../esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_api.c line 353
0x40146e23: gatt_act_write at ../esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_cl.c line 219
0x40146e62: gatt_act_write at ../esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_cl.c line 225
0x40153f91: l2c_link_hci_disc_comp at ../esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_link.c line 379
0x4014293a: btm_sec_rmt_name_request_complete at ../esp-idf/components/bt/host/bluedroid/stack/btm/btm_sec.c line 2991
0x4015d24f: smp_request_oob_data at ../esp-idf/components/bt/host/bluedroid/stack/smp/smp_utils.c line 1619

1mckenna, 2021-06-13:
What version of the esp32 libraries are you using? Same version shown [here?](https://github.com/1mckenna/esp32_iGrill#install-esp32-board-support-with-latest-ble-libs)

jysaloma, 2021-06-14:
[image]
I will now try with older and that version you mentioned on that page.

jysaloma, 2021-06-14:
Seems to work now on that part but throws exception on setting up probes? (I will also double check the mqtt configs)

07:02:17.969 -> [II] Connecting to iGrill Device: d4:81:ca:00:a5:ab
07:02:19.060 -> [II]  - Created client
07:02:19.060 -> [II]  - Connected to iGrill BLE Server
07:02:19.161 -> [II]  - iGrill Pair Status: Paired
07:02:20.059 -> [II]  - Performing iGrill App Authentication
07:02:23.160 -> [II]  - Writing iGrill App Challenge...
07:02:23.674 -> [II]  - Reading iGrill Device Challenge
07:02:24.226 -> [II]  - Writing Encrypted iGrill Device Challenge...
07:02:24.293 -> [II]  - Authentication Complete
07:02:26.553 -> [II] Connecting to MQTT...
07:02:26.553 -> [II] MQTT connection failed: -2
07:02:31.544 -> [II]  - Setting up Battery Characteristic...
07:02:32.055 -> [II]   -- Battery Setup!
07:02:32.531 -> [II]  %% Battery Level: 82%
07:02:32.531 -> [II] Connecting to MQTT...
07:02:32.569 -> [II] MQTT connection failed: -2
07:02:37.555 -> [II]  - Setting up Probes...
07:02:38.043 -> [II]   -- Probe 1 Setup!
07:02:38.043 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
07:02:38.043 -> Core 1 register dump:
07:02:38.043 -> PC      : 0x401e05f3  PS      : 0x00060c30  A0      : 0x800d592c  A1      : 0x3ffcc7a0  
07:02:38.080 -> A2      : 0x00000000  A3      : 0x2b002f24  A4      : 0x3ffc73b8  A5      : 0x2b002f24  
07:02:38.080 -> A6      : 0x3fff4600  A7      : 0x000106ef  A8      : 0x800e800c  A9      : 0x3ffcc740  
07:02:38.080 -> A10     : 0x3ffcc78c  A11     : 0x3fff4600  A12     : 0x00000024  A13     : 0x2b002f24  
07:02:38.080 -> A14     : 0x3ffcc73c  A15     : 0x3ffcc73c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
07:02:38.080 -> EXCVADDR: 0x00000013  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
07:02:38.118 -> 
07:02:38.118 -> ELF file SHA256: 0000000000000000
07:02:38.118 -> 
07:02:38.118 -> Backtrace: 0x401e05f3:0x3ffcc7a0 0x400d5929:0x3ffcc7c0 0x400de2df:0x3ffcc820 0x400f0528:0x3ffcc870 0x40090392:0x3ffcc890
07:02:38.118 -> 
07:02:38.118 -> Rebooting...
07:02:38.118 -> ets Jun  8 2016 00:22:57

-----

PC: 0x401e05f3: BLERemoteCharacteristic::canNotify() at /Users/j/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/libraries/BLE/src/BLERemoteCharacteristic.cpp line 81
EXCVADDR: 0x00000013

Decoding stack results
0x401e05f3: BLERemoteCharacteristic::canNotify() at /Users/j/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/libraries/BLE/src/BLERemoteCharacteristic.cpp line 81
0x400d5929: setupProbes() at /Users/j/OneDrive/Arduino/esp32_iGrill/esp32_iGrill.ino line 309
0x400de2df: loop() at /Users/j/OneDrive/Arduino/esp32_iGrill/esp32_iGrill.ino line 1419
0x400f0528: loopTask(void*) at /Users/j/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/cores/esp32/main.cpp line 23
0x40090392: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

jysaloma, 2021-06-14:
mqtt fixed by enabling the DNS #define USE_CONFIGURABLE_DNS      true

07:36:21.997 -> [II] Connecting to iGrill Device: d4:81:ca:00:a5:ab
07:36:22.989 -> [I][BLEDevice.cpp:614] addPeerDevice(): add conn_id: 0, GATT role: client
07:36:23.163 -> [II]  - Created client
07:36:23.163 -> [II]  - Connected to iGrill BLE Server
07:36:23.339 -> [I][BLEDevice.cpp:253] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
07:36:23.339 -> [II]  - iGrill Pair Status: Paired
07:36:24.151 -> [II]  - Performing iGrill App Authentication
07:36:27.394 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:27.394 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:27.394 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:27.394 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:27.432 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:27.432 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:27.432 -> [II]  - Writing iGrill App Challenge...
07:36:27.969 -> [II]  - Reading iGrill Device Challenge
07:36:28.491 -> [II]  - Writing Encrypted iGrill Device Challenge...
07:36:28.527 -> [II]  - Authentication Complete
07:36:31.071 -> [II] Connecting to MQTT...
07:36:31.142 -> [II] MQTT connected
07:36:31.142 -> [II]  - Setting up Battery Characteristic...
07:36:31.535 -> [II]   -- Battery Setup!
07:36:32.067 -> [II]  %% Battery Level: 80%
07:36:32.067 -> [II]  - Setting up Probes...
07:36:32.067 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:32.067 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
07:36:32.310 -> [II]  * Probe 1 Temp: 25
07:36:32.554 -> [II]   -- Probe 1 Setup!
07:36:32.554 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
07:36:32.554 -> Core 1 register dump:
07:36:32.554 -> PC      : 0x401e235b  PS      : 0x00060c30  A0      : 0x800d5df4  A1      : 0x3ffcc850  
07:36:32.554 -> A2      : 0x00000000  A3      : 0x1a9f178b  A4      : 0x3ffc7468  A5      : 0x1a9f178b  
07:36:32.592 -> A6      : 0x3fff41ac  A7      : 0x000106ef  A8      : 0x800e91d0  A9      : 0x3ffcc7f0  
07:36:32.592 -> A10     : 0x3ffcc83c  A11     : 0x3fff41ac  A12     : 0x00000024  A13     : 0x1a9f178b  
07:36:32.592 -> A14     : 0x3ffcc7ec  A15     : 0x3ffcc7ec  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
07:36:32.592 -> EXCVADDR: 0x00000013  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
07:36:32.628 -> 
07:36:32.628 -> ELF file SHA256: 0000000000000000
07:36:32.628 -> 
07:36:32.628 -> Backtrace: 0x401e235b:0x3ffcc850 0x400d5df1:0x3ffcc870 0x400de5f3:0x3ffcc8d0 0x400f27e0:0x3ffcc920 0x400903ca:0x3ffcc940
07:36:32.628 -> 
------
PC: 0x401e235b: BLERemoteCharacteristic::canNotify() at /Users/j/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/libraries/BLE/src/BLERemoteCharacteristic.cpp line 81
EXCVADDR: 0x00000013

Decoding stack results
0x401e235b: BLERemoteCharacteristic::canNotify() at /Users/j/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/libraries/BLE/src/BLERemoteCharacteristic.cpp line 81
0x400d5df1: setupProbes() at /Users/j/OneDrive/Arduino/esp32_iGrill/esp32_iGrill.ino line 309
0x400de5f3: loop() at /Users/j/OneDrive/Arduino/esp32_iGrill/esp32_iGrill.ino line 1419
0x400f27e0: loopTask(void*) at /Users/j/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/cores/esp32/main.cpp line 23
0x400903ca: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

1mckenna, 2021-06-14:
Awesome looks like I'll just need to add in a check to see if we're setting up a mini device so we only try to setup one probe. 

Can you try commenting out lines 307-324 and see if it runs without crashing?

1mckenna, 2021-06-14:
> Awesome looks like I'll just need to add in a check to see if we're setting up a mini device so we only try to setup one probe.
> 
> Can you try commenting out lines 307-324 and see if it runs without crashing?

I just pushed an updated version to the mini branch that should work (or be close). Can you switch to that branch and then try that version and let me know the results.

jysaloma, 2021-06-14:
I switched to that branch. It seems to work but I not get the devices or entities in Home Assistant via MQTT.
---
23:23:23.846 -> [II]  - Writing iGrill App Challenge...
23:23:24.340 -> [II]  - Reading iGrill Device Challenge
23:23:24.894 -> [II]  - Writing Encrypted iGrill Device Challenge...
23:23:24.929 -> [II]  - Authentication Complete
23:23:27.420 -> [II] Connecting to MQTT...
23:23:27.493 -> [II] MQTT connected
23:23:27.493 -> [II]  - Setting up Battery Characteristic...
23:23:27.916 -> [II]   -- Battery Setup!
23:23:28.430 -> [II]  %% Battery Level: 76%
23:23:28.430 -> [II]  - Setting up Probes...
23:23:28.430 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
23:23:28.430 -> [E][BLERemoteCharacteristic.cpp:282] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
23:23:28.948 -> [II]   -- Probe 1 Setup!
23:23:28.948 -> [II] free heap memory: 77640
23:23:38.941 -> [II] free heap memory: 77776
23:23:40.416 -> [II]  %% Battery Level: 77%% 
23:23:48.941 -> [II] free heap memory: 77484
23:23:58.935 -> [II] free heap memory: 77624
23:24:08.935 -> [II] free heap memory: 75912
23:24:16.191 -> [II]  * Probe 1 Temp: 26
23:24:17.183 -> [II]  * Probe 1 Temp: 25
23:24:18.186 -> [II]  * Probe 1 Temp: 26
23:24:18.932 -> [II] free heap memory: 77484
23:24:22.193 -> [II]  * Probe 1 Temp: 25
23:24:29.251 -> [II] free heap memory: 77464
23:24:36.416 -> [II]  * Probe 1 Temp: 26
23:24:39.248 -> [II] free heap memory: 78744
----
I am using Mosquitto broker version: 6.0.1
I setup the #define MQTT_BASETOPIC           "homeassistant" 
I also enabled the DEBUG logging for mqtt but I not see anything related to iGrill
  logs: 
    homeassistant.components.mqtt: debug

Do I need to create the devices manually or should it create these automatically? Any other idea to try. This probably another issue.

1mckenna, 2021-06-14:
As long as you have MQTT Discovery enabled in Home Assistant it should automatically pick up the device next time home assistant is restarted. I've found its easiest to use the [MQTT Explorer](http://mqtt-explorer.com/) application to connect and verify everything is showing up properly in MQTT.

Can you try installing MQTT Explorer and see what entries are showing up in there for you?

jysaloma, 2021-06-14:
Yes it works now! Sorry my bad. I updated the config file in AP mode to have the base topic also same "homeassistant". 
2021-06-14 23:54:59 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/igrill_d481ca00a5ab/probe_1: b'25'

[image]

[image]

1mckenna, 2021-06-14:
Awesome! Once I get a chance to validate everything still works with the v2 device I'll merge these changes into the main branch  and put your name down in the contributions section as the tester for the mini devices.

Thanks!

1mckenna, 2021-06-15:
iGrill_mini support changes tested and merged into main.

closed 2021-06-15 by 1mckenna