1mckenna/esp32_iGrill #7: Gap Event Crashes on Connection to iGrill 2

ahgray, 2021-07-07:
Not sure what this all means, I followed instructions in readme and am on same board version and everything. Everything seems to work fine until I turn on the igrill 2 then this appears in serial monitor(its truncated) let me know if you need more, but it seems like it just spams it for a while then resets.

`22:01:08.707 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x6
22:01:08.707 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x06 (), length: 16, data: e9a586b322187b8f9e466d2e0f59a7ad
22:01:08.754 -> [D][BLEAdvertisedDevice.cpp:490] setServiceUUID(): - addServiceUUID(): serviceUUID: ada7590f-2e6d-469e-8f7b-1822b386a5e9
22:01:08.754 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
22:01:08.754 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x09 (), length: 18, data: 694772696c6c5f563230322d443437350000
22:01:08.754 -> [D][BLEAdvertisedDevice.cpp:450] setName(): - setName(): name: iGrill_V202-D475
22:01:08.754 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:01:08.800 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:01:08.800 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:01:08.800 -> [D][BLEAdvertisedDevice.cpp:461] setRSSI(): - setRSSI(): rssi: -86
22:01:08.800 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x3
22:01:08.800 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x03 (), length: 2, data: 9ffe
22:01:08.800 -> [D][BLEAdvertisedDevice.cpp:490] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
22:01:08.835 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x16
22:01:08.835 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x16 (), length: 22, data: 9ffe0000000000000000000000000000000000000000
22:01:08.835 -> [D][BLEAdvertisedDevice.cpp:511] setServiceDataUUID(): - addServiceDataUUID(): serviceDataUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
22:01:08.835 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
22:01:08.882 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0xff (), length: 8, data: e0000751ca670dec
22:01:08.882 -> [D][BLEAdvertisedDevice.cpp:438] setManufacturerData(): - manufacturer data: e0000751ca670dec
22:01:08.882 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:01:08.882 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:01:08.882 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:01:08.882 -> [D][BLEAdvertisedDevice.cpp:461] setRSSI(): - setRSSI(): rssi: -49
22:01:08.882 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
22:01:08.923 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
22:01:08.923 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x6
22:01:08.923 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x06 (), length: 16, data: e9a586b322187b8f9e466d2e0f59a7ad
22:01:08.923 -> [D][BLEAdvertisedDevice.cpp:490] setServiceUUID(): - addServiceUUID(): serviceUUID: ada7590f-2e6d-469e-8f7b-1822b386a5e9
22:01:08.923 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
22:01:08.970 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x09 (), length: 18, data: 694772696c6c5f563230322d443437350000
22:01:08.970 -> [D][BLEAdvertisedDevice.cpp:450] setName(): - setName(): name: iGrill_V202-D475
22:01:08.970 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:01:08.970 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:01:08.970 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:01:08.970 -> [D][BLEAdvertisedDevice.cpp:461] setRSSI(): - setRSSI(): rssi: -51
22:01:09.008 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
22:01:09.008 -> [D][BLEAdvertisedDevice.cpp:281] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
22:01:09.008 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x6
22:01:09.008 -> [D][BLEAdvertiseE (86521) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
22:01:09.008 -> E (86521) task_wdt:  - IDLE0 (CPU 0)
22:01:09.008 -> E (86521) task_wdt: Tasks currently running:
22:01:09.008 -> E (86521) task_wdt: CPU 0: BTC_TASK
22:01:09.055 -> E (86521) task_wdt: CPU 1: IDLE1
22:01:09.055 -> E (86521) task_wdt: Aborting.
22:01:09.055 -> abort() was called at PC 0x4019e060 on core 0
22:01:09.055 -> 
22:01:09.055 -> ELF file SHA256: 0000000000000000
22:01:09.055 -> 
22:01:09.055 -> Backtrace: 0x4008f150:0x3ffbe5a0 0x4008f3cd:0x3ffbe5c0 0x4019e060:0x3ffbe5e0 0x40082f89:0x3ffbe600 0x40009217:0x3ffe38c0 0x40007d13:0x3ffe38e0 0x40007c69:0x3ffe3900 0x40008106:0x3ffe3920 0x400f2c3e:0x3ffe39b0 0x400e612a:0x3ffe3a10 0x400eaf85:0x3ffe3a70 0x400e867a:0x3ffe3b50 0x401383d5:0x3ffe3ba0 0x40131e0e:0x3ffe3bc0 0x400903ca:0x3ffe3bf0
22:01:09.055 -> 
22:01:09.055 -> Rebooting...
22:01:09.108 -> ets Jun  8 2016 00:22:57
22:01:09.108 -> 
`

ahgray, 2021-07-07:
Also for what its worth it seems to be happy just periodically scanning, as long as the Igrill is turned off, and prints `22:40:03.510 -> [120412][W][BLEScan.cpp:72] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
22:40:11.461 -> [II] Not connected to an iGrill Device, Skipping connecting to MQTT
22:40:31.488 -> [II] Scanning for iGrill Devices...`
Once you turn on the iGrill 2 it scans once then crashes.

1mckenna, 2021-07-09:
Which version of the esp32 board libs are you using?

ahgray, 2021-07-09:
esp32 by Esprissif (I've Tried Versions 2-alpha1, 1.0.6, and 1.0.5-rc6)
ArduinioJson By Benoit 6.18.0
ESP_DoubleResetDetector By Khoi 1.1.1
ESP_wifimanager by khjoi 1.7.2
LittleFS_esp32 by lorol 1.0.6
PubSubClient by Nick 2.8.0

1mckenna, 2021-07-10:
Can you follow the steps outlined [here](https://github.com/me-no-dev/EspExceptionDecoder) to install the ESPExceptionDecoder. Then go ahead and re-run it and attach the output after the crash.
Also just to confirm you have unpaired your phones/any other devices with the iGrill device so it doesnt try to connect to the iGrill while we are connecting.

1mckenna, 2021-07-10:
I also just confirmed it works ok on my board when using esp32 by Esprissif versoin 1.0.6 as well.

ahgray, 2021-07-10:
Thank you for the help, I just had a batch of new esp32's arrive yesterday so I will try and start fresh on 1.0.6 with a new board, I probably wont be able to get to it till Monday though. As far as unpairing, I've had the  only other device that had connected to the igrill unpaired through the android system menu and then I turned off bluetooth on the phone completely. Should that be sufficient or do i need to do something in the iGrill app?

1mckenna, 2021-07-12:
Unpairing from the Android system menu should be all that's needed.

ahgray, 2021-07-13:
Never used this Exception Decoder before so if i did it wrong please let me know. 
Re-followed your instructions with a brand new Esp32, on 1.0.6 everything works fine untill i turn on the iGrill then it crashes as before, please see exception decode below, thanks.

`07:45:13.487 -> Backtrace: 0x4000c3f5:0x3ffe1ed0 0x4011e925:0x3ffe1ef0 0x40114b1a:0x3ffe1f20 0x4011056a:0x3ffe2230 0x400eb5e7:0x3ffe22f0 0x400eb682:0x3ffe2380 0x400e52ce:0x3ffe2400 0x400d71b1:0x3ffe2440 0x400ea779:0x3ffe2460 0x400e7d06:0x3ffe2540 0x40136815:0x3ffe2590 0x401301da:0x3ffe25b0 0x40090946:0x3ffe25e0`

`
Decoding stack results
0x4011e925: __ssprint_r at ../../../.././newlib/libc/stdio/vfprintf.c line 342
0x40114b1a: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1774
0x4011056a: snprintf at ../../../.././newlib/libc/stdio/snprintf.c line 116
0x400eb5e7: BLEUUID::toString[abi:cxx11]() at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEUUID.cpp line 380
0x400eb682: BLEUUID::equals(BLEUUID) at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEUUID.cpp line 224
0x400e52ce: BLEAdvertisedDevice::isAdvertisingService(BLEUUID) at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEAdvertisedDevice.cpp line 182
0x400d71b1: MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice) at C:\Users\ahgray\Documents\Arduino\2ndtry\esp32_iGrill/esp32_iGrill.ino line 541
0x400ea779: BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEScan.cpp line 125
0x400e7d06: BLEDevice::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEDevice.cpp line 270
0x40136815: btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c line 54
0x401301da: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c line 163
0x40090946: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`

Please note this appears to be different than the original back trace posted in my first comment above, the only thing i changed other than the new board was that arduinojson, pubsubclient, and wifimangager had newer versions during my original attempt and i downgraded them to the versions in your guide for this second attempt. Below please see the decoded output from the original stacktrace.

`Backtrace: 0x4008f150:0x3ffbe5a0 0x4008f3cd:0x3ffbe5c0 0x4019e060:0x3ffbe5e0 0x40082f89:0x3ffbe600 0x40009217:0x3ffe38c0 0x40007d13:0x3ffe38e0 0x40007c69:0x3ffe3900 0x40008106:0x3ffe3920 0x400f2c3e:0x3ffe39b0 0x400e612a:0x3ffe3a10 0x400eaf85:0x3ffe3a70 0x400e867a:0x3ffe3b50 0x401383d5:0x3ffe3ba0 0x40131e0e:0x3ffe3bc0 0x400903ca:0x3ffe3bf0`

`
Decoding stack results
0x4008f150: esp_cpu_reset at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/soc/esp32/cpu_util.c line 49
0x40082f89: set_most_pwr_reg at phy_chip_v7.c line 4040
0x400e612a: BLECharacteristic::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLECharacteristic.cpp line 198
0x400eaf85: BLEService::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEService.cpp line 283
0x400e867a: BLERemoteCharacteristic::readValue[abi:cxx11]() at C:\Users\ahgray\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLERemoteCharacteristic.cpp line 451
0x401383d5: btc_gatts_arg_deep_copy at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c line 121
0x40131e0e: _btc_storage_add_ble_local_key at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_ble_storage.c line 291
0x400903ca: prvInitialiseMutex at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 470
`

1mckenna, 2021-07-15:
Can you change the IGRILL_DEBUG_LVL in the config.h to 2 and then try again and paste the entire content, Serial Log and Decoded backtrace (without your wifi/mqtt info) so I can get a better idea at where it's having an issue.

ahgray, 2021-07-17:
Please see attached log file. The process that generated this log was letting the ESP32 sit idle for a bit, where it scans periodically for an igrill, then I turn the iGrill on, it scans again finds it, crashes twice, then goes into config mode because it looks like a double reset.

[Igrill2Log.log](https://github.com/1mckenna/esp32_iGrill/files/6834969/Igrill2Log.log)

1mckenna, 2021-07-18:
Hmm not seeing anything to helpful in the logs and I don't think it's a BLE library related issue since it's working in the other instances I know of.

Has the ESP been configured to connect to your MQTT instance and Wifi in the portal? In the logs I see this which looks like it's not able to connect to your Wifi. I am not sure if the ESP can handle the Wifi scanning/connecting at the same time as BLE.
```
09:58:39.186 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
09:58:39.186 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
09:58:43.706 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
09:58:43.706 -> [I][WiFiMulti.cpp:114] run(): [WIFI] scan done
09:58:43.706 -> [I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found
09:58:48.466 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
09:58:48.466 -> [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 203 - ASSOC_FAIL
09:58:48.506 -> [E][WiFiMulti.cpp:184] run(): [WIFI] Connecting Failed.
09:58:48.506 -> [D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
09:58:48.506 -> [D][WiFiMulti.cpp:198] run(): [WIFI] start scan
09:58:50.226 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
09:58:54.786 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
09:58:54.826 -> [I][WiFiMulti.cpp:114] run(): [WIFI] scan done
09:58:54.826 -> [I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found

09:58:54.867 -> [E][WiFiMulti.cpp:184] run(): [WIFI] Connecting Failed.
09:58:54.946 -> [D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
09:58:54.946 -> [D][WiFiMulti.cpp:198] run(): [WIFI] start scan
09:58:55.066 -> [WM] WiFi not connected
```

Also just to rule out other random things can you try powering the ESP powering the ESP with a different cable, different USB port, wall adapter if you haven't to see if the behavior is different (i.e. if the ESP is able to connect to the iGrill the blue LED on the iGrill will stay solid blue). 

If none of that gets us anywhere can you try running the example BLE sketch to see if it can pickup the iGrill device without crashing (**File > Examples > ESP32 BLE Arduino and select the BLE_scan example. **).

ahgray, 2021-07-18:
When running the Example BLE sketch it appears to do the same thing. Scan find the iGrill among other things and then just reboot over and over.

Not sure if i needed to format the esp32 again or anything. I just switched over to the example sketch then did verify/upload. 
See attached. 

[abort() was called at PC 0x4015d8a8 on c.log](https://github.com/1mckenna/esp32_iGrill/files/6836908/abort.was.called.at.PC.0x4015d8a8.on.c.log)

1mckenna, 2021-07-22:
How new is the iGrill device? Maybe this one is running a different firmware version. 

1. Install the Weber iGrill App and pair the device to the phone via the app
2. Install the nrfConnect Android App
3. Once in the App Click on the Bonded Tab
4. You should see your iGrill listed, click Connect
5. It should then go through discovery of the services
6. In the listed services can you take a screenshot or attach all the UUIDs/data shown.

closed 2021-10-21 by 1mckenna

ahgray, 2021-11-30:
FYI I know this has been closed for a while as I dropped communication(sorry got deployed), that being said I guessing that in the mean time you've made some changes related to newer Igrill devices, as I just did a git pull, of latest, reflashed and now everything is working. Thanks for all your work.

1mckenna, 2021-12-24:
Awesome, glad to hear its working now for you as well!