1mckenna/esp32_iGrill #8: Retain probe values

Bluhme1, 2021-07-09:
Just a proposal

When I have to restart HA I loose the values of the probes. Would it be possible to change the Arduino code to have these values retained. Where and how, please
And thank you for a great project

1mckenna, 2021-07-09:
I made some updates to make this configurable via changing line 42 in the config.h file to true.
`#define MQTT_RETAIN_TEMP         false //If you want the temperature sensor value to be retained in MQTT. (default: false)`

Can you go ahead and switch to the retainTemp branch, set MQTT_RETAIN_TEMP to true and let me know if everything works as expected.

Thanks

Bluhme1, 2021-07-10:
Dis as instructed (I think). But no luck. After restart of HA the probe entities are listed as unavailable. Battery percent "survives" the restart.

Bluhme1, 2021-07-10:
As you can see from debug information from HA MQTT, the battery_level (the first one) has the Retained flag set whereas the prove 1 sensor (the second one) hasn't got that flag set

Received 17.10.49
QoS: 0, Retained
Payload: 100
Received 17.17.54
QoS: 0
Payload: 100

Received 17.39.08
QoS: 0
Payload: 23
Received 17.39.08
QoS: 0
Payload: 22

1mckenna, 2021-07-10:
When I made the update i forgot i also needed to make a small change in the mqttAnnounce portion of the code as well. I just made the changes and validated on my HA instance that the values are now retained. 
Can you go ahead and do a git pull to grab the latest code then one again make the change on line 42 in the config.h and let me know if it's now working as expected for you.

Bluhme1, 2021-07-10:
Yes, now it works
Thank you very much for your effort and speed (not the least)
I think it's an improvement of a very useful integration. Especially in the summer here in Denmark when grill-evenings are a great pleasure.
Restarting HA may be less often - I hope, but right now it's necessary when upgrading etc.

1mckenna, 2021-07-10:
I just pushed the changes out to the main branch and updated the README documentation. Thanks and Happy Grilling!

closed 2021-07-10 by 1mckenna