1mckenna/esp32_iGrill #9: Add support for propane level sensor (iGrillv3)

eth0up, 2021-08-13:
My iGrillv3 is equipped with a propane level sensor. The official iGrill app reports the propane level. I'm submitting this issue as a feature request to see the same level reporting functionality within esp32_iGrill. 

- Propane Level sensor's BLEUUID is `f5d40001-3548-4c22-9947-f3673fce3cd9s`
- Propane Level sensor has 5 possible values: `100, 75, 50, 25, or 0`
- (Python) the maths used to format the reported level's value: `float(bytearray(self.propane_level.read())[0])*25`
- Implementation in Python, for reference: https://github.com/bendikwa/igrill/pull/24/commits/e6522bf0328554fafbcbc38433d91a491248cd7c

1mckenna, 2021-08-13:
I just pushed an updated version to the propane branch that should work (or be close). Can you switch to that branch and then try that version and let me know the results.

eth0up, 2021-08-14:
> 
> 
> I just pushed an updated version to the propane branch that should work (or be close). Can you switch to that branch and then try that version and let me know the results.

Line #288 in `esp32_iGrill.ino` needs a trailing semicolon for successful compilation in Arduino IDE: https://github.com/1mckenna/esp32_iGrill/blob/dce5b6529a832fe31ee3a79143d82a15ddf46fe6/esp32_iGrill.ino#L288

I uploaded the propane branch sketch to my esp32, but unfortunately I'm not seeing a propane sensor within Home Assistant or MQTT Explorer. I'm attempting to revive my laptop so that I can view the esp logs while being simultaneously within bluetooth range of the igrill (the desktop pc I'm using is not within bt range) . Hopefully the debug output while connected gives us some hints for needed changes. It might be worth noting that my propane tank is currently less than 25% full, so the reported value would be expected as "0".

Thanks for getting started on this enhancement so swiftly!

1mckenna, 2021-08-14:
I made a small update to show if a propane sensor was setup in the systeminfo mqtt topic. Can you try the latest and let me know if in MQTT its showing propane sensor as true or false.

eth0up, 2021-08-15:
> 
> 
> I made a small update to show if a propane sensor was setup in the systeminfo mqtt topic. Can you try the latest and let me know if in MQTT its showing propane sensor as true or false.

MQTT is showing the following for propane sensor:
` "Propane Sensor": "0"`

I'm on software:
`"sw_version": "1.10-000"`

1mckenna, 2021-08-16:
Looks like it's not detecting that characteristic. 

Can you perform the steps below and attach a screenshot of the output here.
1. Install the Weber iGrill App and pair the device to the phone via the app
2. Install the nrfConnect Android App
3. Once in the App Click on the Bonded Tab
4. You should see your iGrill listed, click Connect
5. It should then go through discovery of the services
6. In the listed services can you find the one related to propane level and take a screenshot/attach all the details shown.

eth0up, 2021-08-18:
Here's what I've got from the nrfConnect app:
[image]

Value `0x030000` matches up with 75% fuel level as identified on this git comment:
https://github.com/bendikwa/igrill/issues/12#issuecomment-699583380

Let me know whether there's anything else you'd like me to test. Thanks!

1mckenna, 2021-08-18:
Looks like the propane level is on its own service uuid which is why we weren't able to read that characteristic. I just pushed changes that hopefully should get us some values back. When you get a bit of free time can you pull the latest code and let me know the results.

eth0up, 2021-08-19:
I synced with latest, unfortunately I'm still not seeing the propane sensor in Home Assistant. MQTT Explorer reports `"Propane Sensor": "0"`. 

To double check, I connected via the Weber iGrill App and confirmed that propane level displays as expected:
[image]

1mckenna, 2021-08-19:
Oops realized after you replied that I forgot to change the UUID for the Propane service when I made the updates. Can you re-pull and try again.

eth0up, 2021-08-19:
> 
> 
> Oops realized after you replied that I forgot to change the UUID for the Propane service when I made the updates. Can you re-pull and try again.

MQTT Explorer reports `"Propane Sensor": "0"` after applying the latest change. I confirmed that the app is still displaying the propane level.

1mckenna, 2021-08-19:
Just to confirm you unpaired your phone from the iGrill before testing?

In your latest test is it still reporting the temps to mqtt?

Did you happen to get your laptop running so we can get the debug logs for a closer look at what's happening? If not I'll need to make some updates to throw the debug logs to mqtt.

eth0up, 2021-08-20:
> Just to confirm you unpaired your phone from the iGrill before testing?
>
> In your latest test is it still reporting the temps to mqtt?
>
> Did you happen to get your laptop running so we can get the [...]

I did unpair from my phone before re-testing with esp32. In latest, I was still seeing temperatures and battery reported to MQTT. I also tried forcing `has_propane_sensor=true` but did not see any MQTT output related to propane sensor. 

I just finished getting my laptop up and running and confirmed that I can see the serial monitor output from esp32 as it runs iGrill code.

1mckenna, 2021-08-21:
I just changed how the code is trying to detect if the propane service exists and added in some additional logging. Can you try pulling the most recent and then let me know if its showing in the Serial logs if it detects the propane service or not.

eth0up, 2021-08-23:
After pulling latest the serial logs show that the esp32 successfully connects to the propane service and accurately displays the propane level value - very awesome!
[image]

I'm also seeing messages in MQTT Explorer reporting the propane_level.

1mckenna, 2021-08-23:
Awesome!! Are you also running Home Assistant? If so after a restart of HA did the propane level sensor show up properly there too?

eth0up, 2021-08-23:
> 
> 
> Awesome!! Are you also running Home Assistant? If so after a restart of HA did the propane level sensor show up properly there too?

I am running HA, but the propane sensor does not show up under the igrill device or in the entities. I noticed that the MQTT messages for Battery and Probe (which both show up in HA as sensors under the igrill device) are formatted differently than the propane_level:
[image]

While propane level is accurately reported  in the serial output, it seems that the same value isn't being sent via MQTT.

1mckenna, 2021-08-23:
I just pushed an update that should hopefully fix that issue. You'll likely need to delete the topics via MQTT explorer for things to show up properly. Then if all looks well in MQTT Explorer a restart of HA should get the propane level sensor discovered.

eth0up, 2021-08-23:
After latest update, a propane sensor shows up under the igrill device in HA! The value was reported as "3%", which led me to believe the code for reporting value was missing a multiplication operation.

On my local copy I found and changed [Line#391](https://github.com/1mckenna/esp32_iGrill/commit/9cb7be0a80dd8acbf19ef1e5dd6ff47bd3839200#L391) from `publishPropaneLevel(value);` to `publishPropaneLevel(value*25);` and re-uploaded to the esp32. After this small change Home Assistant now reports a propane sensor with value of 75, which matches my current propane level.

1mckenna, 2021-08-23:
Thank you! I just fixed and pushed that change. If everything else is working well I'll go ahead and push this to the main branch.

eth0up, 2021-08-23:
> 
> 
> Thank you! I just fixed and pushed that change. If everything else is working well I'll go ahead and push this to the main branch.

For good measure I pulled latest and uploaded to the esp32 - it's working very well! Thanks for spending your time and effort to get this feature added.

1mckenna, 2021-08-23:
I just merged this into the main branch and added you to the contributions section in the readme.  Thanks again for the help getting this working and tested.

closed 2021-08-23 by 1mckenna