1mckenna/esp32_iGrill #12: Not connected to an iGrill Device

mwolter805, 2021-09-21:
Having an issue where the ESP isn't connecting to the iGrill. 

Followed the instructions with the following differences. In order to connect to an unbroadcasted SSID, used ESP_WifiManager version 1.7.4 and "don’t use MultiWifi" is enabled. Verified network connectivity via ping test. Made sure all other phones or devices were unpaird. Could it be due to a newer firmware on the iGrill?

iGrill Version: 2.5

```
16:39:56.279 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:39:56.279 -> configsip: 0, SPIWP:0xee
16:39:56.279 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:39:56.316 -> mode:DIO, clock div:1
16:39:56.316 -> load:0x3fff0018,len:4
16:39:56.316 -> load:0x3fff001c,len:1216
16:39:56.316 -> ho 0 tail 12 room 4
16:39:56.316 -> load:0x40078000,len:10944
16:39:56.316 -> load:0x40080400,len:6388
16:39:56.316 -> entry 0x400806b4
16:39:56.953 -> [II] Starting iGrill BLE Client using LittleFS on ESP32_DEV
16:39:56.990 -> [II] ESP_WiFiManager v1.7.4
16:39:56.990 -> [II] ESP_DoubleResetDetector v1.1.1
16:39:56.990 -> {"MQTT_SERVER":"192.168.8.12","MQTT_SERVERPORT":"1883","MQTT_USERNAME":"username","MQTT_PASSWORD":"password","MQTT_BASETOPIC":"igrill","METRIC_DEGREES":true,"NO_MULTI_WIFI":true}[II] 
16:39:56.990 -> Config File successfully parsed
16:39:56.990 -> {"MQTT_SERVER":"192.168.8.12","MQTT_SERVERPORT":"1883","MQTT_USERNAME":"username","MQTT_PASSWORD":"password","MQTT_BASETOPIC":"igrill","METRIC_DEGREES":true,"NO_MULTI_WIFI":true}[II] 
16:39:57.023 -> Config File successfully parsed
16:39:57.023 -> [WM] LoadWiFiCfgFile 
16:39:57.061 -> [WM] OK
16:39:57.061 -> [WM] stationIP = 0.0.0.0 , gatewayIP = 192.168.2.1
16:39:57.061 -> [WM] netMask = 255.255.255.0
16:39:57.061 -> [II] Call connectWiFi in setup
16:39:57.061 -> [WM] ConnectWiFi with :
16:39:57.061 -> [WM] * Using SSID =  ubinot , PW =  password
16:39:57.061 -> [WM] Connecting Wifi...
16:39:57.373 -> [WM] No success, trying next credentials
16:39:57.373 -> [WM] WiFi not connected
16:40:03.012 -> [II] Not connected to an iGrill Device, Skipping connecting to MQTT
16:40:03.012 -> [II] Not connected to an iGrill Device, Skipping connecting to MQTT
16:40:23.006 -> [II] Scanning for iGrill Devices...
16:40:53.030 -> [II] Scanning for iGrill Devices...
16:41:02.995 -> [II] Not connected to an iGrill Device, Skipping connecting to MQTT
16:41:23.031 -> [II] Scanning for iGrill Devices...
16:41:53.007 -> [II] Scanning for iGrill Devices...
```

1mckenna, 2021-09-28:
Looks like it's not detecting the iGrill device. Assume you are testing with the ESP next to the iGrill just to rule out any signal strength issues?
What type of iGrill device do you have and what is the firmware version?

mwolter805, 2021-09-28:
They are about three feet (1m) apart. It's an iGrill 2 and version 2.5.

1mckenna, 2021-09-29:
Can you perform the steps below and attach a screenshot of the output here.

1. Install the Weber iGrill App and pair the device to the phone via the app
2. Install the nrfConnect Android App
3. Once in the App Click on the Bonded Tab
4. You should see your iGrill listed, click Connect
5. It should then go through discovery of the services

mwolter805, 2021-09-29:
Don't have an Android. Does the iPhone app provide the necessary info?

1mckenna, 2021-09-29:
I haven't tried the iOS app, but it looks like it should have the same functionality.

mwolter805, 2021-09-29:
Don't see a Bonded tab in the iPhone app. I'll try to get an android and repeat the steps. Here's the settings from the Grill app. 
[image]
.

mwolter805, 2021-10-08:
Tried installing the iGrill app on a Fire tablet but it was a no go. nrfConnect appeared to work fine though.

Was able to connect to the iGrill with the nrfConnect app on my iPhone and below are the logs. Appears it is showing the services for the device. Does this help?
[text-879104D1CDF6-1.txt](https://github.com/1mckenna/esp32_iGrill/files/7308097/text-879104D1CDF6-1.txt)

1mckenna, 2021-10-11:
That is exactly what I was needing. Can you try pulling the version in the v202 branch and let me know if that fixes it.

mwolter805, 2021-10-12:
Appears to be working... even connected to hidden SSID. Thanks for updating it so quickly. 

Have you thought about porting this to an ESPHome project? Already have another ESP32 with POE for Xiaomi temp sensors and would love to run this off the same device.

[image]
.