1mckenna/esp32_iGrill #13: Weber iGrill V2 not detected

kvoit, 2021-10-14:
I have a Weber iGrill V2 that I recently bought in Germany. Unfortunately, it isn't discovered. In MyAdvertisedDeviceCallbacks::onResult, I added an else statement with a Serial output that tells me that in principle, the iGrill is discoved and haveServiceUUID is true. Thus, I assume the Service UUID of my devise is different from the expected ones.
Any hint on how I find the service UUIDs that are actually advertise, so I can proceed testing?

Thanks for the great work, anyway.

1mckenna, 2021-10-16:
I just added support for the newer iGrillv2 devices a couple days ago, when did you last pull the code? 

If its still not working can you do the following steps and take a screenshot/attach the logs so I can look and see what the UUID your device is using.

1. Install the nrfConnect Android App
2. Once in the App Click on the Bonded Tab
3. You should see your iGrill listed, click Connect
4. It should then go through discovery of the services (this is the info I'll need).

kvoit, 2021-10-16:
It works with the most recent update. Thanks and sorry to bother you!

I wasn't completely up-to-date as I changed the code a bit to be able to flash it with PlatformIO.

gazelle-hub, 2022-02-04:
Not quite sure to export this to a file but here's a screenshot, unless you can tell me?
[image]