1mckenna/esp32_iGrill #14: Errors with FileFS

wbarber69, 2021-12-22:
when i try to compile this I keep getting errors about FileFS  it says something about there being some recommended change in double reset detector

wbarber69, 2021-12-22:
Arduino: 1.8.19 (Windows 10), Board: "NodeMCU-32S, 80MHz, 921600"

In file included from C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:24:

C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\config.h:23: warning: "LED_BUILTIN" redefined

 #define LED_BUILTIN       2

 

In file included from C:\Users\wbarb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:198,

                 from sketch\esp32_iGrill.ino.cpp:1:

C:\Users\wbarb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\variants\nodemcu-32s/pins_arduino.h:16: note: this is the location of the previous definition

 #define LED_BUILTIN LED_BUILTIN

 

In file included from C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:41:

C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h:90:8: warning: #warning Using ESP32 Core 1.0.6 or 2.0.0+ [-Wcpp]

       #warning Using ESP32 Core 1.0.6 or 2.0.0+

        ^~~~~~~

In file included from C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:41:

C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h:94: warning: "FileFS" redefined

       #define FileFS        LittleFS

 

In file included from C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:24:

C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\config.h:13: note: this is the location of the previous definition

 #define FileFS        LITTLEFS

 

C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:102:4: warning: #warning Using DHCP IP [-Wcpp]

   #warning Using DHCP IP

    ^~~~~~~

In file included from C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:41:

C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h: In constructor 'DoubleResetDetector::DoubleResetDetector(int, int)':

C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h:94:29: error: 'LittleFS' was not declared in this scope

       #define FileFS        LittleFS

                             ^~~~~~~~

C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h:148:12: note: in expansion of macro 'FileFS'

       if (!FileFS.begin())

            ^~~~~~

C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h:94:29: note: suggested alternative: 'FileFS'

       #define FileFS        LittleFS

[... 204 lines cut ...]
C:\Users\wbarb\OneDrive\Documents\Arduino\libraries\ESP_DoubleResetDetector\src/ESP_DoubleResetDetector.h:94:29: note: suggested alternative: 'FileFS'

       #define FileFS        LittleFS

                             ^~~~~~~~

C:\Users\wbarb\OneDrive\Documents\esp32_iGrill\esp32_iGrill.ino:1464:8: note: in expansion of macro 'FileFS'

   if (!FileFS.begin(true))

        ^~~~~~

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\wbarb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Error compiling for board NodeMCU-32S.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

1mckenna, 2021-12-24:
Looks like you might be missing some of the libraries. Did you run through [these steps](https://github.com/1mckenna/esp32_iGrill#arduino-ide-setup) to get the arduino IDE setup and install the necessary libraries with the library manager?

wbarber69, 2021-12-24:
I went through and reverted all the libraries to the ones in the pictures, and it worked the first try.  Sorry I’ve not been able to get around to closing this.  Thanks for all the hard work.  I do have some questions about the Lovelace ui custom sensors you have setup.  I can emulate everything and it shows working as it should except for the sensors on the very bottom.  Not sure where to start troubleshooting that. The sensor is called target_alert_temp_probe…. I’ve added all the info you posted in the tutorial and I’m not getting any errors in yamls.

1mckenna, 2021-12-25:
Did you modify the values used in the sensor yaml so they match your igill device sensors? They should be something like sensor.igrill_<igrill bt mac adderss>_probe_#> 

i.e. 
```
- platform: template
  sensors:
    target_alert_temp_probe_1:
      value_template: >-
        {% if (states.sensor.igrill_<igrill bt mac adderss>_probe_1.state | int) >= (states.input_number.igrill_probe_1_target.state | int) %}
          Alert
        {% else %}
          Normal
        {% endif %}
      friendly_name: Probe 1 Target Alert
```

wbarber69, 2021-12-27:
From what I can see apparently homeassistant doesn’t use the sensors.yaml at all.  I had to add it all into my configuration.yaml and then I had to reformat every line because with didn’t like the indentation.  After that I updated and everything seems to have taken.

closed 2022-02-04 by 1mckenna