1mckenna/esp32_iGrill #15: How do you pair an iGrill to your board once flashed?

gazelle-hub, 2022-02-04:
Hi there,

This is really awesome but I can't figure out you pair my iGrill to my esp32 board? There doesn't seem to be any mention of this in your readme file, unless I am missing something?

Thanks in advance for your help.

1mckenna, 2022-02-04:
Unlike when using the iGrill Android/iOS app you won't need to pair the devices. Once the iGrill has connected to the esp32 the blue led on the iGrill device should go solid blue (at least it does on my gen1 iGrillv2 device).

gazelle-hub, 2022-02-04:
So it should just detect that there is an igrill near it and connect?

1mckenna, 2022-02-04:
Correct

gazelle-hub, 2022-02-04:
I've got the iGrill mini and it doesn't seem to connect.

I'm looking at the logs and I can see the my board (M5 Atom Lite) is scanning for devices but it doesn't connect.

I've tried restting the iGrill mini to factory settings but it still doesn't connect.

1mckenna, 2022-02-04:
Can you move the mini next to the esp to make sure its not a BT range issue? Also you may want to double check if you have paired the device to your phone before that you remove the pairing so the phone doesn't try to connect. 

How new is the iGrill mini device (wondering if the new ones have a different service UUIDs similar to the iGrill2 v2 devices)?

gazelle-hub, 2022-02-04:
They're sitting right next to each other while I try and set them up.

I had factory reset the iGrill mini and also removed the app completely. Just redownloaded the app to check and it doesn't have any connnections.

I bought it a couple of weeks ago.

1mckenna, 2022-02-04:
Can you follow the steps [here](https://github.com/1mckenna/esp32_iGrill/issues/13#issuecomment-944920706) and attach the output?

gazelle-hub, 2022-02-04:
Sorry posted on wrong issue.

Here's a screenshot as I'm not sure how to export the info.
[image]

gazelle-hub, 2022-02-05:
Here's the xml file in txt format
[iGrill_MINI02-79B9 - copy.txt](https://github.com/1mckenna/esp32_iGrill/files/8006904/iGrill_MINI02-79B9.-.copy.txt)
.

1mckenna, 2022-02-05:
I just pushed an updated version to the mini_v2 branch that should work now. Can you switch to that branch, flash that version and let me know the results?

gazelle-hub, 2022-02-06:
Thats working great now.

Thanks very much for all of your efforts.

1mckenna, 2022-02-06:
Awesome! iGrill_mini_v2 changes are now merged into the main branch.

Thanks!

closed 2022-02-06 by 1mckenna