1mckenna/esp32_iGrill #16: iGrill mini v2 not showing offline status

gazelle-hub, 2022-02-07:
Hi there,

The iGrill mini v2 that you did has been working well though the issue I've got an the moment is that the igrill mini is still giving a status of online even though it is physically turned off.

It connected and was working but doesn't seem to send an offline message if the device is subsequently turned off.

1mckenna, 2022-02-08:
The status topic is meant for monitoring the connection to the MQTT broker and not the connection to the mini. 
The status topic gets set to online once it establishes a connection to MQTT and only changes to offline if the esp is unplugged or the connection to MQTT is lost.

gazelle-hub, 2022-02-08:
Ok, great thanks.

closed 2022-02-23 by 1mckenna