1mckenna/esp32_iGrill #18: RFC: iGrill vs BlueZ

kratz00, 2022-03-17:
Hi

It looks like my Pulse 2000 does not comply with the Bluetooth specification, for more details see https://github.com/bluez/bluez/issues/316
BlueZ is the Bluetooth protocol stack for Linux. The idea is to find/implement a workaround.

To get a better overview of affected iGrill devices, I kindly ask to provide the complete output of the following command (do not forget to replace the device address) if you want to help:

`btgatt-client -v -d FF:FF:FF:FF:FF:FF`

Additionally the following information might be helpful:

- iGrill device (which one)
- iGrill firmware version
- BlueZ version

Thanks in advance for all your feedback.

kratz00, 2022-04-24:
@PaulAntonDeen Any chance you help me out? Thanks in advance.

1mckenna, 2022-04-25:
Closing as this is not an issue with esp32_iGrill.

Looks like btgatt-client is no longer being included with bluez, but I did run the commands using gatttool and have included the output is below for my iGrillv2 device.
```
$ gatttool --adapter=hci0 --device=D4:81:CA:22:33:48 --primary --characteristics --char-desc
attr handle = 0x0001, end grp handle = 0x0003 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle = 0x0014, end grp handle = 0x0018 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle = 0x0020, end grp handle = 0x002f uuid: 64ac0000-4a4b-4b58-9f37-94d3c52ffdf7
attr handle = 0x0030, end grp handle = 0x0046 uuid: a5c50000-f186-4bd6-97f2-7ebacba0d708
attr handle = 0x0050, end grp handle = 0x0053 uuid: 0000180f-0000-1000-8000-00805f9b34fb
                                                                                          
```

closed 2022-04-25 by 1mckenna

kratz00, 2022-04-26:
@1mckenna Thanks for your feedback.

In hindsight I can not remember why I created this issue in this repository, sorry.

What Linux distribution are you using?
Maybe it is not packing *btgatt-client* (https://github.com/bluez/bluez/blob/master/tools/btgatt-client.c)
*gatttool* is deprecated, see https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=b1eb2c4cd057624312e0412f6c4be000f7fc3617

Nevertheless the tooling choice, could you post a complete log of all services and characteristics,
as matching the characteristics to the services is the interesting part?

kratz00, 2022-04-27:
The issue I am interested is on the ATT protocol level, see the referenced bluez issue in my initial comment.

PaulAntonDeen, 2022-05-02:
Do you know of a mac equivalent utility like btgatt? That would make it a lot easier for me to dump the info you requested.

kratz00, 2022-05-02:
@PaulAntonDeen Unfortunately, I have no experience with Apple hard/software

A quick search came up with using XCode + packetLogger (see https://www.bluetooth.com/blog/a-new-way-to-debug-iosbluetooth-applications/)

beed2112, 2022-05-03:
@kratz00  I wanted to chime in here.  It sounds like the Pulse 2000 is not being picked up by the esp32 program.   This is most likely due to the Pulse 2000 using different UUIDs then the other devices.  

Since this project uses arduino code running on ESP32s is set up to “listen” to defined UUIDs it would be necessary to identify the UUIDs used by the Pulse 2000.

Here is a closed issue that discusses the steps needed to identify the UUIDs the device uses.

https://github.com/1mckenna/esp32_iGrill/issues/13

Happy sniffing

PaulAntonDeen, 2022-05-03:
@kratz00 I got a fresh copy of rasbian on a RPi4 and I can install applications as needed. So talk me through what you want me to share/dump?
Gatttool is working so what command line options do you need?

kratz00, 2022-05-03:
@beed2112
Thanks
You are correct some UUIDs are different. As I have access to a Pulse 2000 I have already investigate them.
The problem I am investigating at the moment is a possible bug in the firmware, see bluez/bluez#316 for more information.

@PaulAntonDeen
Thanks for setting up a debug environment :+1: 

Replace *FF:FF:FF:FF:FF:FF* with the MAC of your grill.
```
gatttool -b FF:FF:FF:FF:FF:FF --primary 
gatttool -b FF:FF:FF:FF:FF:FF --characteristics
gatttool -b FF:FF:FF:FF:FF:FF --char-desc
```

PaulAntonDeen, 2022-05-03:
```
gatttool -b 70:91:8F:7A:CB:09 --primary 
attr handle = 0x0001, end grp handle = 0x0003 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle = 0x0010, end grp handle = 0x0014 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle = 0x0015, end grp handle = 0x001d uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle = 0x0020, end grp handle = 0x002f uuid: 64ac0000-4a4b-4b58-9f37-94d3c52ffdf7
attr handle = 0x0030, end grp handle = 0x004b uuid: 7e920000-68dc-41c7-943f-518b278cea88
attr handle = 0x0080, end grp handle = 0x0083 uuid: 0000180f-0000-1000-8000-00805f9b34fb
attr handle = 0x0090, end grp handle = 0x00b3 uuid: 6c910000-58dc-41c7-943f-518b278ceaaa
```

```
gatttool -b 70:91:8F:7A:CB:09 --characteristics
handle = 0x0002, char properties = 0x10, char value handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0011, char properties = 0x02, char value handle = 0x0016, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0013, char properties = 0x02, char value handle = 0x0018, uuid = 00002a01-0000-1000-8000-00805f9b34fb
handle = 0x0016, char properties = 0x02, char value handle = 0x0017, uuid = 00002a24-0000-1000-8000-00805f9b34fb
handle = 0x0018, char properties = 0x02, char value handle = 0x0019, uuid = 00002a25-0000-1000-8000-00805f9b34fb
handle = 0x001a, char properties = 0x02, char value handle = 0x001b, uuid = 00002a29-0000-1000-8000-00805f9b34fb
handle = 0x001c, char properties = 0x02, char value handle = 0x001d, uuid = 00002a27-0000-1000-8000-00805f9b34fb
handle = 0x0021, char properties = 0x02, char value handle = 0x0022, uuid = 64ac0001-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0023, char properties = 0x08, char value handle = 0x0024, uuid = 64ac0002-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0025, char properties = 0x02, char value handle = 0x0026, uuid = 64ac0003-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0027, char properties = 0x08, char value handle = 0x0028, uuid = 64ac0004-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0029, char properties = 0x08, char value handle = 0x002a, uuid = 64ac0005-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x002b, char properties = 0x02, char value handle = 0x002c, uuid = 64ac0006-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x002d, char properties = 0x12, char value handle = 0x002e, uuid = 64ac0007-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0031, char properties = 0x0a, char value handle = 0x0032, uuid = 06ef0001-2e06-4b79-9e33-fce2c42805ec
handle = 0x0033, char properties = 0x12, char value handle = 0x0034, uuid = 06ef0002-2e06-4b79-9e33-fce2c42805ec
handle = 0x0036, char properties = 0x0a, char value handle = 0x0037, uuid = 06ef0003-2e06-4b79-9e33-fce2c42805ec
handle = 0x0038, char properties = 0x12, char value handle = 0x0039, uuid = 06ef0004-2e06-4b79-9e33-fce2c42805ec
handle = 0x003b, char properties = 0x0a, char value handle = 0x003c, uuid = 06ef0005-2e06-4b79-9e33-fce2c42805ec
handle = 0x003d, char properties = 0x12, char value handle = 0x003e, uuid = 06ef0006-2e06-4b79-9e33-fce2c42805ec
handle = 0x0040, char properties = 0x0a, char value handle = 0x0041, uuid = 06ef0007-2e06-4b79-9e33-fce2c42805ec
handle = 0x0042, char properties = 0x12, char value handle = 0x0043, uuid = 06ef0008-2e06-4b79-9e33-fce2c42805ec
handle = 0x0045, char properties = 0x0a, char value handle = 0x0046, uuid = 06ef0009-2e06-4b79-9e33-fce2c42805ec
handle = 0x0047, char properties = 0x12, char value handle = 0x0048, uuid = 06ef000a-2e06-4b79-9e33-fce2c42805ec
handle = 0x004a, char properties = 0x0a, char value handle = 0x004b, uuid = 06ef000b-2e06-4b79-9e33-fce2c42805ec
handle = 0x0081, char properties = 0x12, char value handle = 0x0082, uuid = 00002a19-0000-1000-8000-00805f9b34fb
handle = 0x0091, char properties = 0x12, char value handle = 0x0092, uuid = 6c910001-58dc-41c7-943f-518b278ceaaa
handle = 0x0094, char properties = 0x0a, char value handle = 0x0095, uuid = 6c910002-58dc-41c7-943f-518b278ceaaa
handle = 0x0097, char properties = 0x02, char value handle = 0x0098, uuid = 6c910003-58dc-41c7-943f-518b278ceaaa
handle = 0x009b, char properties = 0x0a, char value handle = 0x009c, uuid = 6c910004-58dc-41c7-943f-518b278ceaaa
handle = 0x009e, char properties = 0x12, char value handle = 0x009f, uuid = 6c910005-58dc-41c7-943f-518b278ceaaa
handle = 0x00a1, char properties = 0x12, char value handle = 0x00a2, uuid = 6c910006-58dc-41c7-943f-518b278ceaaa
handle = 0x00a5, char properties = 0x12, char value handle = 0x00a6, uuid = 6c910007-58dc-41c7-943f-518b278ceaaa
handle = 0x00a8, char properties = 0x12, char value handle = 0x00a9, uuid = 6c910008-58dc-41c7-943f-518b278ceaaa
handle = 0x00ab, char properties = 0x0a, char value handle = 0x00ac, uuid = 6c910009-58dc-41c7-943f-518b278ceaaa
handle = 0x00af, char properties = 0x0a, char value handle = 0x00b0, uuid = 6c91000a-58dc-41c7-943f-518b278ceaaa
handle = 0x00b2, char properties = 0x0a, char value handle = 0x00b3, uuid = 6c91000b-58dc-41c7-943f-518b278ceaaa
```

```
gatttool -b 70:91:8F:7A:CB:09 --char-desc
handle = 0x0001, uuid = 00002800-0000-1000-8000-00805f9b34fb
handle = 0x0002, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0010, uuid = 00002800-0000-1000-8000-00805f9b34fb
handle = 0x0011, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x0012, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0013, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x0014, uuid = 00002a01-0000-1000-8000-00805f9b34fb
[... cut ...]
handle = 0x00a8, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00a9, uuid = 6c910008-58dc-41c7-943f-518b278ceaaa
handle = 0x00aa, uuid = 00002902-0000-1000-8000-00805f9b34fb
handle = 0x00ab, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00ac, uuid = 6c910009-58dc-41c7-943f-518b278ceaaa
handle = 0x00af, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00b0, uuid = 6c91000a-58dc-41c7-943f-518b278ceaaa
handle = 0x00b2, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00b3, uuid = 6c91000b-58dc-41c7-943f-518b278ceaaa
```

kratz00, 2022-05-03:
@PaulAntonDeen
Thank you very much for the data.
Does you grill also run on firmware version 0.07-rc1? (I think Weber never released any firmware updates so far)

PaulAntonDeen, 2022-05-04:
Yes same firmware