1mckenna/esp32_iGrill #19: A lot of compile errors

bertdeelman, 2022-06-13:
Hi,

First thanks for your great work! Really nice. 
 
I followed the clear instructions and add the Arduino libraries. But I receive a lot of compile errors about redefining. 

Is there a solution for this? It was mentioned before but that topic was closed. 

Regards,
Bert

bertdeelman, 2022-06-14:
Hi Gerben,

I made some progress. It seems to be that the LITTLEFS library is now part of the core Arduino. I use the IDE v1.8.19 macOS version. So there is no need to add that library to the sketch. LITTLEFS is renamed to LittleFS. After those changes I can upload it to ESP32 board. It starts, portal ok, WiFi Ok. But some weird error message after Performing iGrill App Authentication. Guru Meditation Error: Core  0 panic'ed (Double exception). 

I need to figure out what is happening here and if I can debug. 

Groet,
Bert

GerbenBakker, 2022-06-16:
Hi Bert, I recall having the same/similar issue. Not too much of an arduino expert myself so I will not be able to guide you any further than sharing that by including the exact library versions (including ESP board) i was able to compile the code without any modification. My ealier attempt to chage LITTLEFS to LittleFS resulted in same error that you currently have over the serial interface. Happy to send you my compiled binary (for DFRobot FireBeetle ESP8266 IoT Development Board - Wi-Fi - ).

westado, 2022-06-23:
I have this fault  if i compile

'custom_MQTT_SERVER_LEN' was not declared in this scope

westado, 2022-06-23:
Is it possible that someone send me the bin file?

westado, 2022-06-28:
> > Is it possible that someone send me the bin file?
> > [esp32_iGrill.ino.esp32.zip](https://github.com/1mckenna/esp32_iGrill/files/8995556/esp32_iGrill.ino.esp32.zip)
> > Hope this helps

Thanks it works!!

1mckenna, 2022-08-04:
Sorry for the delayed response, but this should be fixed now. I've tested and confirmed it working on my end using the latest versions of all the libraries and the built in littlefs library (I updated the README to list the versions i tested with).

Can you try again and let me know if it resolves your issue?

andreq, 2022-08-14:
Not the OP, but tried with latest libraries and latest code, seem to be all good. 
Also seem to fix the random case where my esp32 would just decide to go offline after a while 👍

1mckenna, 2022-08-17:
Thanks for testing and letting me know it worked for you as well. I am going to go ahead and close this issue now.

closed 2022-08-17 by 1mckenna

bjhiltbrand, 2022-11-06:
This is not working in my instance. I have installed all dependencies using the versions listed in the Readme, yet I still get the following error when attempting to compile:

C:\Workspaces\esp32_iGrill\esp32_iGrill.ino:35:105: fatal error: LittleFS.h: No such file or directory