1mckenna/esp32_iGrill #20: Propane Service: electrical details needed

LukasQ, 2022-08-24:
Dear all

Unfortunately my iGrill3 was sold without a fuel integration here in Europe. I think mainly bc of the different tanks.
Still, I want to have a propane level indicator in HA and it seems like you can't buy them.
Could one please check the iGrill signals on that iGrill3 jack, so I can hack and 3D print an open source solution?
Spoiler: I tried load cells already. Even though they seem pretty cool, they're also pretty temperature sensitive, which is to me an issue an the BBQ ;)

Cheers, Lukas

1mckenna, 2022-09-05:
I don't have an iGrill3/haven't seen one in person to know enough about how the propane level sensor portion works. Maybe someone with an iGrill3 can reply and provide some more info on how that part works.

LukasQ, 2022-09-05:
I recreated one with a spring and a sliding potentiometer and updated your code with an ADC and a some basic math. Should do the trick, at least more precise than a 1/4 scale.
[image]

closed 2022-12-29 by 1mckenna