1mckenna/esp32_iGrill #23: iGrill3 Device Does Not Automatically Appear

r0gu3Ag3n7, 2022-09-14:
> **After you have configured the esp32_iGrill Client and have it connected to the same MQTT Broker as your Home Assistant instance the device should automatically appear.**

At this point my device does not automatically appear in the Home Assistant devices.  Within the MQTT config settings, if I listen to "#" then I see the igrill messages.  I can also see the info in MQTT explorer.  But even after restarting the Mosquitto integration and the entire Home Assistant itself, the device is not detected.  

Any help would be greatly appreciated.  I seem to have the MQTT broker setup correctly as well as the ESP32, but I'm not sure why the device doesn't show up.

adrianpele, 2022-09-14:
Exactly the same for me too

adrianpele, 2022-09-14:
Change to your desired base MQTT Topic
If you are using Home Assistant and want to take advantage of MQTT Autodiscovery you need to set this to your mqtt autodiscoervy prefix.
(Home Assistant Default: **homeassistant**)

r0gu3Ag3n7, 2022-09-14:
> Change to your desired base MQTT Topic
> If you are using Home Assistant and want to take advantage of MQTT Autodiscovery you need to set this to your mqtt autodiscoervy prefix.
> (Home Assistant Default: **homeassistant**)

Just to clarify, this is set in the "config.h" file prior to flashing the ESP32?  My default in there is set to "igrill" but I need to change it to "homeassistant" to support Auto Discovery?  Or can this also be done in the config portal when you are configuring the post-flashed ESP32 for wifi/etc?

I think the language could be a little more clear in this section "Default Configuration Portal Settings", because both igrill and homeassistant are referred to as defaults in different sections.

r0gu3Ag3n7, 2022-09-15:
I got it!  Thanks!!!

closed 2022-09-15 by r0gu3Ag3n7

AndersNielsen85, 2022-09-28:
@r0gu3Ag3n7 mind providing your solution? - I can't seem to get it autodiscovered even when setting to "home assistant"

r0gu3Ag3n7, 2022-09-28:
> @r0gu3Ag3n7 mind providing your solution? - I can't seem to get it autodiscovered even when setting to "homeassistant"

I'm not exactly sure WHICH worked.  In the config.h file I set the MQTT base topic to "homeassistant" and re-flashed the ESP32.  I ALSO had to set it during initial set up in the web interface as noted here [https://github.com/1mckenna/esp32_iGrill/blob/main/README.md#default-configuration-portal-settings](url)

Hope this helps!

1mckenna, 2022-09-28:
You should set it in the config portal to **homeassistant**
If you already flashed the ESP and configured it to connect to Wifi then whatever settings you chose in the config portal will be retained as they get stored on the device and persist through a reflash unless you wipe the partition. This is likely why you have changed the config.h and its still publishing to the igrill base topic.