1mckenna/esp32_iGrill #25: Integration with ESPHome

Daedilus, 2022-11-20:
As my coding skills are limited, is there a way to integrate this with ESPHome? It would allow someone to use the low power bluetooth options already combined and then allow the ESP32 device to be used for multiple applications without having it as a standalone device that only does iGrill.

sn3ak, 2022-11-30:
@mwolter805 posted an ESPHome configuration [here](https://github.com/esphome/issues/issues/3319).
It may be unnecessary to integrate this with ESPHome. At the time that configuration was posted mwolter805 was having issues with the battery reporting though.
Having just stumbled across this myself, I will hopefully be trying it later this evening.

@mwolter805, If you were able to solve this issue with a configuration change would you be so kind to share it with us?

sn3ak, 2022-12-01:
So far battery state works fine. I am unable to get the probes to report temperature.

1mckenna, 2022-12-28:
In order to get the temperature probe values you have to authenticate/pair with the device. If ESPHome supports that then it should be possible.

closed 2022-12-29 by 1mckenna