1mckenna/esp32_iGrill #28: Status Online with iGrill Turned Off

robferre, 2023-04-13:
this is awesome - thanks all
i have noticed though when i turn off the igrill v2 that the status shows "online" and stays like that with all probes retained... 
im watching the mqtt flow of info via mqtt explorer and dont see the status change
then when i turn it back on - the status stays but the probe info updates
cant get it to get to offline when i turn the igrill off
is there something i need to do differently?

1mckenna, 2023-04-14:
The status MQTT topic is used to keep track of whether or not ESP32 device is connected to the MQTT server. If you unplug the ESP32 it will change to offline. 

If you are wanting the temp probes to reset to a blank value you can just unplug the temp probes from the iGrill before turning it off.

robferre, 2023-04-14:
Yeah right so it's not so much if the esp has a Bluetooth connection. More
if the esp is on or not...
There isn't a sensor to detect Bluetooth connection or not? (Ie igrill is
on or off)

1mckenna, 2023-05-25:
I normally unplug/plug my esp in when using the iGrill, but it does make sense to add in a sensor for when the iGrill is connected/disconnected. I'll work on getting the code updated and let you know when there is a branch ready for you to test and confirm its working correctly.

1mckenna, 2023-05-31:
I just pushed a the [statusUpdates](https://github.com/1mckenna/esp32_iGrill/tree/statusUpdates) branch that should resolve this for you. 

The code now 
- Creates a separate esp32 sensor with some info system attributes
- Creates a sensor to track connectivity with the iGrill
- Probe connection/disconnection now will automatically add/remove the probes from mqtt as they are connected/disconnected making the HA UI experience nicer. (only applies if you use the default temp probe retain setting).

When you get a chance try pulling the branch and let me know how it works on your end.