1mckenna/esp32_iGrill #29: Last initialized probe not functioning

MSmeets94, 2023-05-12:
Hi, thank you for this great project!

I am using an iGrill 2. I am having an issue, where the last probe which is initialized in the setup does not seem to function. I first tought something was wrong with my 4th probe. After discovering it is working fine in the Android app and investigating further, i noticed that when switching around the intialization, suddenly probe 3 stopped working. 
```         
          probe4TempCharacteristic = iGrillService->getCharacteristic(PROBE4_TEMPERATURE);
          if(probe4TempCharacteristic->canNotify())
          {
            probe4TempCharacteristic->registerForNotify(notifyCallback);
            IGRILLLOGGER("  -- Probe 4 Setup!",1);
          }
          probe2TempCharacteristic = iGrillService->getCharacteristic(PROBE2_TEMPERATURE);
          if(probe2TempCharacteristic->canNotify())
          {
            probe2TempCharacteristic->registerForNotify(notifyCallback);
            IGRILLLOGGER("  -- Probe 2 Setup!",1);
          }
          probe3TempCharacteristic = iGrillService->getCharacteristic(PROBE3_TEMPERATURE);
          if(probe3TempCharacteristic->canNotify())
          {
           probe3TempCharacteristic->registerForNotify(notifyCallback);
           IGRILLLOGGER("  -- Probe 3 Setup!",1);
          }
```
When starting the ESP / iGrill without any probes connected, we can also see the callback is not invoked:

```
[II] MQTT connected
[II]  - Setting up Battery Characteristic...
[II]   -- Battery Setup!
[II]  %% Battery Level: 80%% 
[II]  %% Battery Level: 80%
[II]  - Setting Device Temp Units...
[II]   ** Temperature Units: METRIC
[II]  - Setting up Probes...
[II]   -- Probe 1 Setup!
[II]   -- Probe 4 Setup!
[II]   -- Probe 2 Setup!
[II]   -- Probe 3 Setup!
[II] free heap memory: 50404
[II]  ! Probe 1 Disconnected!
[II]  ! Probe 2 Disconnected!
[II]  ! Probe 4 Disconnected!
```

Any idea what could be wrong?

antraynl, 2023-05-15:
Having the same issue with my iGrill2.
When executing "setupBatteryCharacteristic();" after "setupProbes();", all probes are functioning but the battery level will not be updated anymore.
Board used: Olimex ESP32-POE-ISO rev. K

1mckenna, 2023-05-25:
> Hi, thank you for this great project!
>
> I am using an iGrill 2. I am having an issue, where the last probe which is initialized in the setup does not seem to function. I first tought something was [...]

So just to confirm all the probes work individually or up to three at a time? What versions of the libraries and board are you using?

1mckenna, 2023-05-25:
> Having the same issue with my iGrill2. When executing "setupBatteryCharacteristic();" after "setupProbes();", all probes are functioning but the battery level will not be updated anymore. Board used: Olimex ESP32-POE-ISO rev. K

Does the battery level ever report a status?

Battery level will only get reported on initial connect and then again when the value changes (i.e. battery percentage drops).

antraynl, 2023-05-26:
> Does the battery level ever report a status?
>
> Battery level will only get reported on initial connect and then again when the value changes (i.e. battery percentage drops).

All probes are connected.
First what is logged with first battery characteristics, then probe characteristics:

> 19:40:00.597 -> [II] MQTT connected
> 19:40:00.597 -> [II]  - Setting up Battery Characteristic...
> 19:40:00.662 -> [II]   -- Battery Setup!
> 19:40:00.695 -> [II]  %% Battery Level: 57%
> 19:40:00.695 -> [II]  - Setting Device Temp Units...
> 19:40:01.816 -> [II]   ** Temperature Units: METRIC
> 19:40:01.816 -> [II]  - Setting up Probes...
> 19:40:01.882 -> [II]   -- Probe 1 Setup!
> 19:40:01.916 -> [II]   -- Probe 2 Setup!
> 19:40:01.982 -> [II]   -- Probe 3 Setup!
> 19:40:02.013 -> [II]   -- Probe 4 Setup!
> 19:40:02.013 -> [II] free heap memory: 61004
> 19:40:02.937 -> [II]  * Probe 1 Temp: 23
> 19:40:03.961 -> [II]  * Probe 1 Temp: 22
> 19:40:09.990 -> [II]  * Probe 2 Temp: 22
> 19:40:10.023 -> [II]  * Probe 3 Temp: 22
> 19:40:10.980 -> [II]  * Probe 3 Temp: 23
> 19:40:12.034 -> [II] free heap memory: 60824
> 19:40:21.954 -> [II]  %% Battery Level: 56%% 

As you can see, probe 4 is not reporting the temperature although connected and working properly.
Battery level change is reported (from 57% to 56%)

Now what is logged with first probe characteristics, then battery characteristics:

> 19:47:51.523 -> [II] Connecting to MQTT...
> 19:47:51.556 -> [II] MQTT connected
> 19:47:51.556 -> [II]  - Setting Device Temp Units...
> 19:47:52.644 -> [II]   ** Temperature Units: METRIC
> 19:47:52.677 -> [II]  - Setting up Probes...
> 19:47:52.743 -> [II]   -- Probe 1 Setup!
> 19:47:52.777 -> [II]   -- Probe 2 Setup!
> 19:47:52.808 -> [II]   -- Probe 3 Setup!
> 19:47:52.840 -> [II]   -- Probe 4 Setup!
> 19:47:52.840 -> [II]  - Setting up Battery Characteristic...
> 19:47:52.907 -> [II]   -- Battery Setup!
> 19:47:52.939 -> [II]  %% Battery Level: 55%
> 19:47:52.939 -> [II] free heap memory: 59108
> 19:47:53.530 -> [II]  * Probe 3 Temp: 22
> 19:47:53.564 -> [II]  * Probe 4 Temp: 22
> 19:47:58.578 -> [II]  * Probe 1 Temp: 22
> 19:47:58.578 -> [II]  * Probe 2 Temp: 23
> 19:47:59.567 -> [II]  * Probe 1 Temp: 23
> 19:47:59.600 -> [II]  * Probe 3 Temp: 21
> 19:47:59.600 -> [II]  * Probe 4 Temp: 21

As you can see now is that probe 4 is reporting temperature and battery level is only reported on initial connection, it will not report any changes after that.

MSmeets94, 2023-05-26:
All probes, and battery are working individually. I do notice the same behaviour as @antraynl when switching around the probe and battery initialization; If i initialize my probes before the battery, then all 4 probes seem to be working (=> sending updates). However, my battery does not update anymore. 

I also tried removing one BLE characteristic from initializing (one which is not the last), after which all initialized probes/ temps are working, including the last.

This leads me to believe that somehow a limit is triggered on subscriptions or something, which leads to the last BLE characteristic  which is subscribed on to not function properly. 

Please see my attached screenshot for the libraries. I used the same minor versions you've tested with. Only difference i used ESP_DoubleResetDetector 1.3.2. 

[image]

dsabathi, 2023-05-27:
same issue here with an original iGrillv2 (before Weber bought iGrill).
Even when I only connect probe4, I don't get any value send via MQTT.
Debug shows all 4 probes:
[II]  - Setting Device Temp Units...
[II]   ** Temperature Units: METRIC
[II]  - Setting up Probes...
[II]   -- Probe 1 Setup!
[II]   -- Probe 2 Setup!
[II]   -- Probe 3 Setup!
[II]   -- Probe 4 Setup!

Battery level is transmitted:
II]  %% Battery Level: 31%% 

And as soon as I connect any other probe, I get the value transmitted:
[II]  * Probe 2 Temp: 27

When I disconnect the iGrill from the esp_32 and connect it to my iPhone, the weber app sees all 4 probes and I get the correct values..

1mckenna, 2023-05-30:
Good news, I did some testing with this on the weekend and was able to reproduce the issue you are seeing using the latest version of the esp32 board 2.0.9, but reverting to the last version I listed in the docs (2.0.4) everything started working properly again. I still need to run through the other versions to see which one introduced the issue.

Can you please try changing the version of ESP32 to version 2.0.4 then reflash and let me know if this also resolves the issue for you?

MSmeets94, 2023-05-31:
Thank you for checking. 
To be honest i totally skipped (and also read) over selecting the correct board version. Maybe it was due to the new Arduino UI? I can't remember..  Anyway, I can confirm downgrading to 2.0.4 fixes the problem. I did also run trough the other versions and it seems version 2.0.7 is causing the problem, altough 2.0.6 is also already starting to behave differently ( callbacks are coming in before the initialization is complete, meaning something changed in the interrupts / timing i guess)

After finding out the problem lies in the version of the esp32 boards, i digged trough the github repo and found this issue:
https://github.com/espressif/arduino-esp32/issues/8015 which seems to correspond with that same 2.0.7 version. 

I'll post in that issue over there link to this issue, but this doesnt seem to be a bug on your side. I will revert back to 2.0.4 for now and enjoy everything working again :-).

closed 2023-06-06 by 1mckenna