1mckenna/esp32_iGrill #31: Not as much an issue..

cptcave, 2023-06-07:
Forgive me as I am a bit of a noob.  I have followed the instructions up to the merging of the files and have hass talking to my esp32 and my iGrill v2 is reporting data back to hass through the esp32.  The automatically recognized mqtt device tends to drop connection at random and is unstable. I suspect that is what the rest of the instructions are to address.  I understand appending the files, but the instruction to replace with the iGrill mac address is not clear.  In none of the code in any of the files can I find a mac address to replace. I see examples of several preconfigured existances of esp32_iGrill but not in ALL files as the instruction indicates.  Is there a place in every file to place the mac? Also, what is the current working version of hassio that supports this integration?

Thank you for the great work!

1mckenna, 2023-06-08:
For the instability issue please check and ensure the esp32 board library version you are using is 2.0.4.

Probably the easiest way to get your mac address now that you have connected the iGrill to HA once is to open your HA Dashboard then Settings > Devices and Services > Devices Tab > filter for igrill. Listed there you should see a device that starts with igrill_. The part after the _ is the devices mac address. Now with your device name you should just be able to replace my iGrill device name (igrill_d481ca223348) in the configs with yours.

There used to be a manual MQTT sensor you needed to configure to monitor the iGrill's status, but with the latest changes I made this sensor now gets automatically configured and setup via MQTT Autodiscovery. Is that what you are referring to? 

This will work with any Home Assistant installation since it's just using MQTT to send over the device and sensor information.

cptcave, 2023-06-08:
Thank you for responding. I had used 2.0.4 but at some point i was asked to do an update, so i did automatically.   Ill change it back.

I was referring to these instructions

[Image]
 It was here I could not find the place to change the Mac address.

Again, this is great. I saw bendikwa's version but it required too much coding and work to get it going.  This is so far the easiest way to make this happen.

Thanks,
Kevin

cptcave, 2023-06-08:
Ok, duh, I see now about the mac.  I did not realize the mac was the number following igrill. Even though you specifically note it.  Now the format is wrong on my dashboard.  I only get battery levels and probe temps.  Everything else has errors. The state bar is completely blank. Everything else says " entity not available".  You list all of the mqtt topics. Am I supposed to list those somehwere?

closed 2023-08-14 by 1mckenna