1mckenna/esp32_iGrill #35: Error: 'WiFi_AP_IPConfig' does not name a type; did you mean 'WM_Config'?

anderswernersson, 2023-09-26:
Hello,

I get the following error when trying to compile:

esp32_iGrill:122:1: error: 'WiFi_AP_IPConfig' does not name a type; did you mean 'WM_Config'?
 WiFi_AP_IPConfig WM_AP_IPconfig;
 ^~~~~~~~~~~~~~~~
 WM_Config
esp32_iGrill:123:1: error: 'WiFi_STA_IPConfig' does not name a type
 WiFi_STA_IPConfig WM_STA_IPconfig;
 ^~~~~~~~~~~~~~~~~
esp32_iGrill:735:27: error: variable or field 'initAPIPConfigStruct' declared void
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                           ^~~~~~~~~~~~~~~~
esp32_iGrill:735:27: error: 'WiFi_AP_IPConfig' was not declared in this scope
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:735:27: note: suggested alternative: 'WM_Config'
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                           ^~~~~~~~~~~~~~~~
                           WM_Config
esp32_iGrill:735:45: error: 'in_WM_AP_IPconfig' was not declared in this scope
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                                             ^~~~~~~~~~~~~~~~~
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:735:45: note: suggested alternative: 'WM_config'
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                                             ^~~~~~~~~~~~~~~~~
                                             WM_config
esp32_iGrill:742:28: error: variable or field 'initSTAIPConfigStruct' declared void
 void initSTAIPConfigStruct(WiFi_STA_IPConfig &in_WM_STA_IPconfig)
                            ^~~~~~~~~~~~~~~~~
esp32_iGrill:742:28: error: 'WiFi_STA_IPConfig' was not declared in this scope
esp32_iGrill:742:47: error: 'in_WM_STA_IPconfig' was not declared in this scope
 void initSTAIPConfigStruct(WiFi_STA_IPConfig &in_WM_STA_IPconfig)
                                               ^~~~~~~~~~~~~~~~~~
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:742:47: note: suggested alternative: 'WM_config'
 void initSTAIPConfigStruct(WiFi_STA_IPConfig &in_WM_STA_IPconfig)
                                               ^~~~~~~~~~~~~~~~~~
                                               WM_config
esp32_iGrill:753:28: error: variable or field 'displayIPConfigStruct' declared void
 void displayIPConfigStruct(WiFi_STA_IPConfig in_WM_STA_IPconfig)
                            ^~~~~~~~~~~~~~~~~
esp32_iGrill:753:28: error: 'WiFi_STA_IPConfig' was not declared in this scope
esp32_iGrill:762:17: error: variable or field 'configWiFi' declared void
 void configWiFi(WiFi_STA_IPConfig in_WM_STA_IPconfig)
                 ^~~~~~~~~~~~~~~~~
esp32_iGrill:762:17: error: 'WiFi_STA_IPConfig' was not declared in this scope
esp32_iGrill:735:27: error: variable or field 'initAPIPConfigStruct' declared void
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                           ^~~~~~~~~~~~~~~~
esp32_iGrill:735:27: error: 'WiFi_AP_IPConfig' was not declared in this scope
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:735:27: note: suggested alternative: 'WM_Config'
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                           ^~~~~~~~~~~~~~~~
                           WM_Config
esp32_iGrill:735:45: error: 'in_WM_AP_IPconfig' was not declared in this scope
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                                             ^~~~~~~~~~~~~~~~~
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:735:45: note: suggested alternative: 'WM_config'
 void initAPIPConfigStruct(WiFi_AP_IPConfig &in_WM_AP_IPconfig)
                                             ^~~~~~~~~~~~~~~~~
                                             WM_config
esp32_iGrill:742:28: error: variable or field 'initSTAIPConfigStruct' declared void
 void initSTAIPConfigStruct(WiFi_STA_IPConfig &in_WM_STA_IPconfig)
                            ^~~~~~~~~~~~~~~~~
esp32_iGrill:742:28: error: 'WiFi_STA_IPConfig' was not declared in this scope
esp32_iGrill:742:47: error: 'in_WM_STA_IPconfig' was not declared in this scope
 void initSTAIPConfigStruct(WiFi_STA_IPConfig &in_WM_STA_IPconfig)
                                               ^~~~~~~~~~~~~~~~~~
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:742:47: note: suggested alternative: 'WM_config'
 void initSTAIPConfigStruct(WiFi_STA_IPConfig &in_WM_STA_IPconfig)
                                               ^~~~~~~~~~~~~~~~~~
[... 58 lines cut ...]
esp32_iGrill:1532:25: error: 'WM_STA_IPconfig' was not declared in this scope
   initSTAIPConfigStruct(WM_STA_IPconfig);
                         ^~~~~~~~~~~~~~~
/Users/werand/Documents/Arduino/esp32_iGrill/esp32_iGrill.ino:1532:25: note: suggested alternative: 'WM_config'
   initSTAIPConfigStruct(WM_STA_IPconfig);
                         ^~~~~~~~~~~~~~~
                         WM_config
esp32_iGrill:1532:3: error: 'initSTAIPConfigStruct' was not declared in this scope
   initSTAIPConfigStruct(WM_STA_IPconfig);
   ^~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Using library ArduinoJson at version 6.19.4 in folder: /Users/werand/Documents/Arduino/libraries/ArduinoJson 
Using library WiFi at version 2.0.0 in folder: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/WiFi 
Using library PubSubClient at version 2.8 in folder: /Users/werand/Documents/Arduino/libraries/PubSubClient 
Using library ESP_WiFiManager at version 1.3.0 in folder: /Users/werand/Documents/Arduino/libraries/ESP_WiFiManager 
Using library WebServer at version 2.0.0 in folder: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/WebServer 
Using library DNSServer at version 2.0.0 in folder: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/DNSServer 
Using library BLE at version 2.0.0 in folder: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/BLE 
Using library LittleFS at version 2.0.0 in folder: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/LittleFS 
Using library FS at version 2.0.0 in folder: /Users/werand/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/libraries/FS 
Using library ESP_DoubleResetDetector at version 1.3.1 in folder: /Users/werand/Documents/Arduino/libraries/ESP_DoubleResetDetector 
exit status 1
'WiFi_AP_IPConfig' does not name a type; did you mean 'WM_Config'?

1mckenna, 2023-09-28:
Looks like you are missing some of the required libraries. 

Try installing the libraries mentioned in the Readme and let me know if you are still having an issue.

anderswernersson, 2023-10-05:
I removed and reinstalled all my libraries, then it worked. Thank you for the help and sorry for the confusion!

closed 2023-10-05 by anderswernersson