1mckenna/esp32_iGrill #36: No Network connection after initial configuration

AtrejaAlwinnen, 2024-02-08:
Hi,
I have the following problem:
I can flash my ESP and connect to it’s AP. However, when i set up my network settings with fixed IP, the ESP show up in my WiFi network with a DHCP IP-Adress, but I cannot connect to it with my browser. A connection to the MQTT broker is not established, just like there is no connection to my Pulse 2000 grill.

Board:
ESP32-WROOM-32D

Arduino IDE:
2.2.1

Board Manager:
esp32 2.0.11 (2.0.4 not available)

Librarys:
ArduinoJson 6.19.4
ESP-DoubleRestDetector 1.3.0
ESP_WIFIManager 1.10.2
LittleFS_esp32 1.0.6
PubSubClient 2.8

1mckenna, 2024-02-17:
From what I recall the esp32 wont connect to MQTT for the first time until it sees and connects to an iGrill device. 

Currently the code doesn't support the Pulse2000 grill but looks like it wouldn't be too hard to add if you're willing to do the testing for it.

AtrejaAlwinnen, 2024-02-17:
I now use an ESPHome-iGrill program to connect to my Pulse 2000. But of course I can carry out further tests with a second ESP32. However, I don't have much experience in ESP and Arduino programming. But I can certainly download newer versions, try them out and, if necessary, post logs or something similar here.

ufondu88, 2024-10-12:
I have 2 questions: 

1. How can we connect the iGrill to the flashed ESP32?
2. After connecting the ESP32 to Wifi, how can I access the web portal of the ESP32?

1mckenna, 2024-10-12:
>     1. How can we connect the iGrill to the flashed ESP32?
 The iGrill just needs to be powered on within range of the esp32

>     2. After connecting the ESP32 to Wifi, how can I access the web portal of the ESP32?
If you need to change a setting you can access the web portal again by pressing the reset button 2x in a short time. This causes the Double-Reset detector code to trigger and launch the portal.

> Also, as I am not very familiar with MQTT, how can I ensure that the ESP32 is properly connected to the MQTT broker? I do not see any logs in the broker

I use a tool called [MQTT Explorer](http://mqtt-explorer.com/), once you connect to your MQTT broker you should be able to view the igrill device live.